opencax/GSoC #1: Improve DXF import and export

t-paul, 2019-02-07:
## Outline

Currently OpenSCAD uses it's own limited parser to read DXF files (2D-only) which does not support things like curves/splines. Using an external library should provide better support for both import and export of DXF files.

## Details

The main focus for the change is the import of DXF files as the import currently fails to read various categories of DXF files exported by other tools. Right now a complex preparation is needed to ensure drawings created in programs like Inkscape can be imported correctly in OpenSCAD. Using an existing external library with complete DXF support should make the process much easier.

Importing more complex drawings will need some integration with OpenSCAD features, e.g. the handling of [$fn/$fa/$fs](https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/Other_Language_Features#.24fa.2C_.24fs_and_.24fn) variables that define how curves are converted into the internal polygon/mesh data structures.

As the libraries normally support both reading and writing of DXF files, the export could be changed to use the selected library too. This is not highly critical as the export is based on a known set of internal data structures but would help to unify the code base. In case the library supports 3D DXF files, it would be nice to add support for that.

Topics that need to be solved
* Research options of usable libraries
* Investigate support for 2D and 3D import/export
* Select library and integrate into OpenSCAD
* Add test cases to verify the new export/import functionality
* Update build system to include the new library into installers

## Expected Outcome

* OpenSCAD supports reading of 2D DXF files exported by other programs (e.g. Inkscape) without special preparation before the export (e.g. convert curves to straight lines)
* Optional: Support for reading and writing 3D DXF

## Project Properties

### Skills

* Programming language is C++
* Understand and use APIs from external libraries
* Integrate new libraries into the build system for the 3 supported platforms
* Add test cases with files using the new features to allow regression testing

### Difficulty

Easy

## Additional Information

Known libraries:
* libdxfrw - http://sourceforge.net/projects/libdxfrw/
* DxfLib - http://www.ribbonsoft.com/de/what-is-dxflib
* Dime - https://bitbucket.org/Coin3D/dime
* BRL-CAD - see http://sourceforge.net/p/brlcad/code/HEAD/tree/brlcad/trunk/src/conv/dxf/ also there's currently a project going on to provide file-formats/converters as separate library

avinash14022002, 2021-03-24:
Wanted to work on this issue?
Can anyone guide how to start?

brlcad, 2021-03-24:
@avinash14022002 One way to get started could be to read up on the previous GSoC project that worked on this task: https://github.com/openscad/openscad/issues/3044

Of course, read the links in the description too including the current issue tracking integration: openscad/openscad#3006

t-paul, 2021-03-24:
@avinash14022002 in addition, check [Getting Started](https://github.com/openscad/openscad/wiki/Google-Summer-of-Code---Getting-Started) for a short summary of general steps.

Yinchu-Liu, 2021-03-25:
I want to join this project, but where does the source code locate in?

avinash14022002, 2021-03-25:
@brlcad sir and @t-paul sir,
sir could you guide through the project and good first issue..!!!

brlcad, 2021-03-25:
> I want to join this project, but where does the source code locate in?

@Yinchu-Liu did you see the links provided?  The openscad/openscad#3006 issue is tracking the latest code changes for integration into OpenSCAD.  openscad/openscad#3044 has a great overview of the whole thing including other links to source code (e.g., pre-merge and indep library setup).

Yinchu-Liu, 2021-03-25:
> @Yinchu-Liu did you see the links provided? The [openscad/openscad#3006](https://github.com/openscad/openscad/pull/3006) issue is tracking the latest code changes for integration into OpenSCAD. [...]

Got it. Thank you!

brlcad, 2021-03-25:
> @brlcad sir and @t-paul sir,
> sir could you guide through the project and good first issue..!!!

@avinash14022002 if you need that much help already even with the detailed links provided, then GSoC may not be a great fit.  Conversion is technically challenging and typically needs someone able and willing to figure things out, not be guided on every step.  I suggest compile and run openscad first, compile the dxf code, and find something you can improve (however minor) to have a discussion on that.

avinash14022002, 2021-03-25:
@brlcad thank you sir..!!!

Yinchu-Liu, 2021-03-26:
Just want to make the problem more clear, what we should do is improve BRL-CAD code , i.e. dxf-g.c, to OpenSCAD so that OpenSCAD can also import outside dxf documents? Or rather, what we should implement is:

- Reading dxf-g.c in BRL-CAD.
- Add similar code with dxf-g.c in OpenSCAD but in a C++ style to use C++ allocate memory, say the name should be dxf-openscad.cpp.
- However, in 2019 project, the coversion of 2D splines and Bezier still has issues, we should fix the problem with the guidance with mathematical formulus.

Is my understanding clear? Or anything else we can add, although I think those could fit the workload for a suitable time slot?

Yinchu-Liu, 2021-03-26:
> Just want to make the problem more clear, what we should do is improve BRL-CAD code , i.e. dxf-g.c, to OpenSCAD so that OpenSCAD can also import outside dxf documents? Or rather, what we should [...]

Sorry for my misunderstanding, we should work on dxf.cc instead of creating one called dxf-openscad.cpp.

brlcad, 2021-03-26:
Others can speak to it better but as I understand it, the goal of this project is not to improve or do anything with BRL-CAD code.  It's to complete the DXF import support for OpenSCAD, which had origins and previous work that involved dxf-g.c from BRL-CAD.  It would be awesome if the dxf parsing library worked on for this effort continued to be made reusable so that it "could" be re-integrated back into BRL-CAD for dxf support, but the focus is what's in the description -- completing the integration into OpenSCAD and cleaning up the dxf library code.

Yinchu-Liu, 2021-03-26:
> Others can speak to it better but as I understand it, the goal of this project is not to improve or do anything with BRL-CAD code. It's to complete the DXF import support for OpenSCAD, which had [...]
Yes, that's also what I am thinking. Let me look into the OpenSCAD dxf file and get hands dirty. Thank you!

aykalkan, 2021-03-30:
Just a little correction. You wrote difficulty as medium in project description, but tagged as easy. This project is definitely not easy.

closed 2022-02-07 by t-paul