opencax/GSoC #2: Improve OpenGL rendering

t-paul, 2019-02-07:
## Outline

The current rendering infrastructure is scattered around with various bits partially belonging to external packages (CGAL, OpenCSG). Also, we're currently using immediate-mode OpenGL in some places. This is suspected to degrade rendering performance of large models.

## Details

The existing code should be refactored and rewritten, preferably by implementing or using an existing rendering/scenegraph framework. In addition to better OpenGL support and performance, care should be taken to use modern OpenGL so that the new code code be ported to GLES2 in the future.
Supporting GLES2 opens up much more possibilities like WebGL and native support for ARM Linux platforms that usually only provide GLES2 implementation.

## Expected Outcome

* Cleaner rendering code
* Better support and better responsiveness for bigger models
* Optional: Support for ARM platforms using GLES2 (e.g. Raspberry Pi)

## Project Properties

### Skills

* Programming language is C++
* Knowledge of OpenGL and GLES v2
* Ability to read and readiness to refactor existing code

### Difficulty

Medium to High

## Additional information

See https://github.com/openscad/openscad/wiki/Information-About-Other-Frameworks

### Related issues on Github

* Speed up OpenGL 3d mesh drawing significantly by batching whole polyheda [#675](https://github.com/openscad/openscad/issues/675)
* Improve GUI responsiveness for complex objects in preview mode [#418](https://github.com/openscad/openscad/issues/418)
* Display of 2-dim objects [#362](https://github.com/openscad/openscad/issues/362)
* build problems with libgles2 [#292](https://github.com/openscad/openscad/issues/292)
* Progressive display of rendered components [#64](https://github.com/openscad/openscad/issues/64)
* Provide antialias / multisampling settings for the 3d view [#1769](https://github.com/openscad/openscad/issues/1769)

### Further functionality which could be enabled by this

* GUI object inspection (e.g. highlight bbox, size of imported STLs)
* Pick object in 3D window and highlighting in the code
* Pick object in the code and highlight in the 3D window

Jedi18, 2019-12-28:
Hi @t-paul , I'd like to know if this project would be considered in GSoC 2020, if BRL-CAD gets selected.
This looks like an interesting project, and starting now would give me enough time to get acquainted with the code base and master the prerequisites.

t-paul, 2019-12-28:
Yes, if OpenSCAD will be selected for GSoC, this topic will certainly be included. There's probably quite some research that could be done before actually diving into the details.

Jedi18, 2019-12-28:
That's great, I'll start looking into it now then. If I have any doubts I hope I can message on the IRC channel?

t-paul, 2019-12-28:
Yes, discussion on IRC is definitely recommended.

closed 2022-02-07 by t-paul