opencax/GSoC #3: Improve Text Editor

t-paul, 2019-02-07:
## Outline

The integrated text editor improved quite a bit recently but there's still a number of features missing that are often requested and would make editing bigger projects easier.

## Details

This project is about adding more IDE style features to the OpenSCAD editor. Some of those features have already a prototype implementation so those would require review of the existing code continue from there.

Examples for useful features:
* Multi-file editing
* Context sensitive help / autocomplete while typing code
* Simple changing of values / numbers via mouse
* Better integration with the parser (error reporting, jump to definitions, ...)

The main challenge is to select just a small number of features but implement those in a way that really improves the usage. Ideally this involves making different prototypes and including feedback from the user community.

## Expected Outcome

* Improved text editing experience for OpenSCAD users

## Project Properties

### Skills

* Programming language is C++
* GUI programming with the Qt framework

### Difficulty

Medium

## Additional Information

Links:
* Prototype for Autocomplete/Call-Tips https://github.com/openscad/openscad/pull/905
* Patches for Abbreviations feature http://forum.openscad.org/Abbreviations-td17583.html

AkashAli506, 2019-03-14:
Hi @t-paul,
I am interested in improving the text editor of OpenSCAD. Can you please help me how to start?

t-paul, 2019-03-14:
A good start could be just running OpenSCAD with some example code that is included in the normal installation and have a look at other Tools/IDEs. Some features that would be nice to have that are present in other applications are listed above, but those are just guidelines.
One important step is to select a small number of features (or just one bigger topic) and proposing a plan how to improve/add this to the OpenSCAD editor.
For more interactive discussion you can join the IRC channel on freenode linked above.

afolabisamguy, 2019-03-21:
Am interested in improving this text editor,  i have a list of ideas, how do i get the opencax source code

t-paul, 2019-03-21:
@afolabisamguy This project idea is for OpenSCAD which has the main repo at https://github.com/openscad/openscad

closed 2021-02-13 by t-paul