opencax/GSoC #4: Add support for exporting models in STEP format

t-paul, 2019-02-07:
## Outline

The STEP format is widely used in the industry to transfer CAD data between different systems. Currently OpenSCAD does not support STEP import or export. Adding STEP export would open up a number of new usecases or simplify the workflow as no external conversion tools are needed to convert to STEP. This includes the design of 3D models for other CAD tools, e.g. for KiCAD where STEP models are used to render 3D representations of PCBs. Other use cases are for manufacturing where sometimes only STEP files are accepted as input, e.g. for CNC milling services.

## Details

The main focus of this project is to get the ground work done for exporting more detailed models, as opposed to just exporting the fully rendered single mesh which is the normal case right now.

Topics that need to be solved
* Research options of usable libraries
* Investigate what type of STEP files are accepted as input by various tools
* Select library and integrate into OpenSCAD
* Implement base functionality to export single meshes
* Add test cases to verify the new export functionality
* Update build system to include the new library into installers
* Prototype how more advanced models can be exported

## Expected Outcome

* OpenSCAD supports exporting single meshes as STEP
* (optional) Understanding/Plan of how to support additional features supported by STEP

## Project Properties

### Skills

* Programming language is C++
* Understand and use APIs from external libraries
* Integrate new libraries into the build system for the 3 supported platforms
* Add test cases with files using the new features to allow regression testing

### Difficulty

Hard

## Additional Information

Known libraries:
* StepCode - http://stepcode.org/  (https://github.com/stepcode/stepcode)
* OpenCASCADE - https://www.opencascade.com/

hxadogar, 2025-03-21:
Hi,

I came across this issue and wanted to check its current status. Has STEP export support been implemented, or is it still open for development?

If there is still work left, I would love to contribute. Let me know where I could start.

hxadogar, 2025-03-21:
Hi @t-paul,

I came across this issue and wanted to check its current status. Has STEP export support been implemented, or is it still open for development?

If there is still work left, I would love to contribute. Let me know where I could start.

t-paul, 2025-03-21:
It is still open.

KeithSloan, 2025-03-21:
Surely an OpenSCAD STEP exporter could only support Meshes within a STEP file.

My understanding is that STEP can support Meshes and that some importers may not recognise.

When people are asking for an OpenSCAD  STEP exporter do they understand the limitation?

t-paul, 2025-03-22:
> Surely an OpenSCAD STEP exporter could only support Meshes within a STEP file.

This is wrong and you know that. Why claim such things if the FreeCAD import of OpenSCAD files proves otherwise?

We may start with just exporting meshes as first step so the scope and timeline is kept reasonable. But that's a decision to be made in the project proposal.

KeithSloan, 2025-03-22:
"This is wrong and you know that. Why claim such things if the FreeCAD import of OpenSCAD files proves otherwise?"

The FreeCAD OpenSCAD importer, struggles with file that use hull, minkowski and polyhedron ( which is basically creating a mesh).

The importer works by using OpenSCAD to (under the covers )create a csg file. It then parses the csg with ply and creates an appropriate/equivalent Brep Shape. It struggles with hulls and  Minkowski and Sebastian Hoogen one of the two initial developers tried his best to handle these by passing requests to OpenSCAD to produce a mesh.

Now if a similar approach  is take by some native OpenSCAD code, then I agree it could be done more efficiently than passing back and forth between OpenSCAD and FreeCAD but will not escape the limitations and it would loose the parameters that define Shapes.

Now I notice that libraries like BSOL build up functions from basic ones that create polyhedrons, which is understandable but the end result when passed as a csg to FreeCAD (or what ever) ends up a "Mesh' which does not matter to OpenSCAD but again means in a STEP export a Mesh for the created objects.

I just think that anybody working on this as part of a Google SOC  should understand this or magically come up with a way of
addressing the limitation.

KeithSloan, 2025-03-22:
I think it might be possible to implement a hull operation with Brep Shapes but even starting with just two objects that is probably a Masters or Phd project.

One could then recursively reduce <n> shapes to one, if the algorithm for two could not be applied to <n> all at once.

Would be of interest to not just OpenSCAD maybe a joint SOC project?

Adelkazzaz, 2025-03-23:
Hello @t-paul , I have been preparing for this project for a while to write a proposal for this project.
I am in the stage of understanding the project and making it possible for us to make it up and running and understand the export mechanisms currently. Is it still available? Because I see some discussions about it

t-paul, 2025-03-23:
@Adelkazzaz Yes, it's available and with the application period starting tomorrow, it's the perfect time to discuss - ideally talking in person via the #openscad IRC channel. As the discussion above shows, one critical topic is trimming down the scope in regard to difficulty and planned features. Expecting a full export (e.g. with a similar full feature-set as the import in FreeCAD is very much unrealistic).

If we find some topic crossing between OpenSCAD and FreeCAD (or one of the other projects), I'm be happy to discuss too. At this point, I don't have a clear view on that. The proposal described above is written from the standpoint of OpenSCAD.

closed 2026-02-03 by t-paul