opencax/GSoC #5: Survey of CSG algorithms

t-paul, 2019-02-07:
## Outline

Constructive solid geometry ([CSG](https://en.wikipedia.org/wiki/Constructive_solid_geometry)) is one of the two core modeling techniques provided by OpenSCAD. Currently, 3D CSG operations are handled using the [CGAL](http://www.cgal.org) library. 2D CSG operations are handled using the [Clipper library](http://www.angusj.com/delphi/clipper.php).

There is a large number of papers about the topic of CSG and related algorithms. This project is intended to provide an overview about available algorithms, their properties and a thorough comparison.

## Details

* Survey existing CSG algorithms
* Compare properties of the algorithms
* Implement prototypes that could be tested inside the OpenSCAD framework
* Evaluate how more advanced modeling techniques could be realized with each algorithm (e.g. minkowski sums, convex hulls)

## Future Opportunities
* Smoothing/fillets of intersection edges
    * http://implicitcad.org
    * http://www.iquilezles.org/www/articles/smin/smin.htm

## Expected Outcome

* A document/paper containing the CSG algorithm survey and how it relates to OpenSCAD-specific issues.
* One or more prototype implementations integrated with OpenSCAD to demonstrate possible improvements

## Project Properties

### Skills

* Programming language is C++
* Admiration for math and algorithms
* Good understanding of computational geometry
* Ability to read and understand research papers
* Systematic strategy to collect and organize information

### Difficulty

Medium to High

## Additional information

### CSG Research

* http://liris.cnrs.fr/mepp/
* MEPP: [Exact and Efficient Booleans for Polyhedra](http://liris.cnrs.fr/Documents/Liris-4883.pdf)
* Gilbert Bernstein and Don Fussell: [Fast, Exact, Linear Booleans](http://www.gilbertbernstein.com/project_boolean.html)
* Sebastian Steuer: [Methods for Polygonalization
of a Constructive Solid Geometry Description in Web-based Rendering Environments](http://www.en.pms.ifi.lmu.de/publications/diplomarbeiten/Sebastian.Steuer/DA_Sebastian.Steuer.pdf)

### Other papers

* 3D Delaunay Triangulation on the GPU (http://www.comp.nus.edu.sg/~tants/gdel3d.html)

### Existing libraries or software
* [Cork](https://github.com/gilbo/cork) - by Gilbert Bernstein
* [libigl](http://libigl.github.io/libigl/tutorial/tutorial.html#booleanoperationsonmeshes) - [blog](http://www.alecjacobson.com/weblog/?tag=libigl)
* [Carve](https://code.google.com/p/carve) (see also pyPolyCSG) ([on github](https://github.com/folded/carve) / [a fork on github](https://github.com/VTREEM/Carve))
* [OpenCASCADE](http://www.opencascade.org) (see also [OCE](https://github.com/tpaviot/oce), [PythonOCC](http://www.pythonocc.org), [Cadmium](https://github.com/jay3sh/cadmium), [CADQuery](https://github.com/dcowden/cadquery) )
* [CSGTool](https://github.com/sshirokov/csgtool)
* [csg.js](https://github.com/evanw/csg.js)
* [csgjs-cpp](https://github.com/dabroz/csgjs-cpp)
* [ThreeCSG](https://github.com/chandlerprall/ThreeCSG)
* [Babylon.js CSG](https://github.com/BabylonJS/Babylon.js/blob/master/src/Mesh/babylon.csg.ts)
* [CSG-BSP](https://github.com/mkkellogg/CSG-BSP)

### Related software and other resources
* http://www.plasm.net

AnisaAisha, 2020-03-19:
Hello @t-paul, I'm interested in working on this project for GSoC. Could you please guide me where to start?

t-paul, 2020-03-19:
As the application period is already open, I'd suggest starting a draft proposal after reviewing the starting points above. Think about how to split the time into a) review/research and what criteria would be useful and b) coding example programs. For GSoC, the coding part should be a big part of cause.
With an initial draft it's easier to discuss in a more interactive way on our IRC channel.

closed 2022-02-07 by t-paul