opencax/GSoC #6: Offline documentation

t-paul, 2019-02-07:
## Outline

The [user manual](http://en.wikibooks.org/wiki/OpenSCAD_User_Manual) for OpenSCAD is maintained at [Wikibooks](http://wikibooks.org/). This is a nice solution as everybody with a Wikipedia account can help improving it. The drawback is that it's only available online. The "Print Version" accessible there needs special work and duplicates part of the information.

It would be very useful if generating an offline version that could be shipped with OpenSCAD could be automated.

## Details

Wikipedia allows the export of data via a simple interface (https://en.wikipedia.org/wiki/Help:Export)

This is probably mainly intended to be the source for backups and/or for transferring data from one installation to another. There's big list of converters around, most seem to target the import into another wiki software but some will convert to other file formats already. (https://www.mediawiki.org/wiki/Alternative_parsers)

The manual export is easily done by calling https://en.wikibooks.org/wiki/Special:Export and entering "OpenSCAD_User_Manual" into the "Add pages from category". This exports all the text content into one big XML file.

The ideal option would be an automated way to convert this XML file into different formats. As first step, to make it usable as off-line documentation, the target would probably be either static HTML that could be shipped with OpenSCAD or maybe PDF.

It needs investigating if that actual programming is required, or if the existing tools are maybe already sufficient for that. So the specific tasks could be one of

1. write a script that does all the conversion and image downloading

2. write a simple script to just run various existing tools

3. go though the wiki pages of the manual and adjust some of the
   text so it works nicely with some existing tool

4. use a tool or write a script to scrape the HTML representation from Wikibooks

5. something else, we did not see yet...

Ideally that should run on Linux so it could be integrated into one of the automatic build servers (e.g. Travis). A manual solution that is just started now an then would also be a nice thing.

### Challenges:

* Investigate current status of available converters
* Optional: Integrate the solution into the OpenSCAD build process

## Expected Outcome

* An offline user manual that can be included in OpenSCAD releases.

## Project Properties

### Skills

* Programming language is python (or some other widely used scripting language... make a suggestion!)

### Difficulty

Easy

## Additional information

css911, 2019-03-04:
Hello... @t-paul . I am Chetan from India. I am interested in contributing to this idea can you help me,like how can i further explore the idea..?

t-paul, 2019-03-04:
If you have not read it yet, the official GSoC [student guide](https://google.github.io/gsocguides/student/) is a good place to start learning how the overall process works. Some more organisation specific details can be found in the process section of the [OpenCAx website](https://opencax.github.io/).

As for the project itself, I'd suggest to look at the different possible ways outlined above. For more interactive discussion join the [OpenSCAD IRC channel](https://webchat.freenode.net/?channels=openscad) (people are in widely different time zones so it can take a while to catch someone, I'm adding the nick names on IRC to the project descriptions...).

css911, 2019-03-07:
Thanks @t-paul :)

RomitKumar, 2019-03-11:
Hi @t-paul, I am Romit Kumar, 3rd year undergraduate from India. I am interested in solving the issue. I would like to know whether the issue is taken or not.  I also want to which is the best place to discuss about my approach towards solving the issue. Thank you :)

t-paul, 2019-03-11:
Hi @RomitKumar, the best place for discussion is our IRC channel (see link above).

RomitKumar, 2019-03-12:
Is there some issue, maybe some low effort or easy one, that I can solve to give a sample of my work

t-paul, 2019-03-12:
How about openscad/openscad#2729?

ixil, 2019-06-11:
https://github.com/openscad/openscad/issues/2206#issuecomment-500889806

shruti-tanpure, 2021-04-10:
Hey! I would like to work on this issue.

t-paul, 2021-04-10:
That's still possible, with the deadline approaching fast and other students having quite a head start, you may need to get into the topic in a very focused way.

shruti-tanpure, 2021-04-14:
Okay! Thank you @t-paul

closed 2021-08-28 by t-paul