opencax/GSoC #7: Port the GUI to C++

t-paul, 2019-02-09:
## Outline

The GUI is currently coded in Perl using wxWidgets and our goal is to have it ported to C++. The wxWidgets API is almost identical between Perl and C++, so it's an easy task. There are a couple things where we use the dynamic features of Perl which are trickier and interesting to port.

## Project Properties

### Skills

* C++/wxWidgets

### Difficulty

Easy

## Additional information

AhmedAminSE, 2019-03-10:
I am interested ! can i have more information and what exactly do you want to see in the proposal ?

lordofhyphens, 2019-03-10:
@AhmedAminSE
Overall scope of the work is as follows:
* Extend the current work in https://github.com/slic3r/Slic3r/tree/cppgui 
* Implement the profile editors.

lordofhyphens, 2019-03-10:
We will want you to submit a pull request for one of the following https://github.com/slic3r/Slic3r/labels/Low%20Effort so that we have an example of your work.

sagen98, 2019-03-30:
I want to participate in GSoC 2019 in your organization. So I want to know what will be the format of the proposal and also what will be the file extension type of the proposal.

lordofhyphens, 2019-04-01:
@sagen98 See the rules for GSoC. We're going to want a code sample in the form of a pull request for one of the existing Low Effort issues in the slic3r/Slic3r  tracker.