opencax/GSoC #8: Port the SVGNest library to C++

t-paul, 2019-02-09:
## Outline

The SVGNest library provides an algorithm for polygon nesting. It's written in JavaScript and a C++ port of it would improve Slic3r's auto-arrange feature. 

## Project Properties

### Skills

* C++

### Difficulty

Easy

## Additional information

* slic3r/Slic3r#3237

Sagnik-Chaudhuri, 2019-03-01:
What should be my next set of procedures to take part in this project for GSOC 2019?

t-paul, 2019-03-01:
A good list of places to go is 1) the process section of the [OpenCAx website](https://opencax.github.io/), 2) the official GSoC [student guide](https://google.github.io/gsocguides/student/) and 3) the [Slic3r IRC channel](https://webchat.freenode.net/?channels=slic3r) for contacting people.

lordofhyphens, 2019-03-02:
Also asking around in the issue itself.

lordofhyphens, 2019-03-10:
@Sagnik-Chaudhuri The other thing you should do is look through the Slic3r issue tracker and do one of the "Low Effort" issues and submit a pull request (so that we have a sample of your work).

lordofhyphens, 2019-03-10:
https://github.com/slic3r/Slic3r/labels/Low%20Effort