opencax/GSoC #15: BCF support

yorikvanhavre, 2019-02-26:
This page is dedicated  to the description of the [Google Summer of Code](https://www.freecadweb.org/wiki/Google_Summer_of_Code_2019) project idea of adding BCF support to FreeCAD.

## Outline

The [BIM Collaboration Format](https://www.buildingsmart.org/bim-collaboration-format-explained/) (or BCF) is a file format aimed at coordinating the work of several participants in a [ Building Information Modeling (BIM)](https://en.wikipedia.org/wiki/Building_information_modeling) project. The format specifications are open and [maintained by BuildingSMART](http://www.buildingsmart-tech.org/specifications/bcf-releases), the same consortium behind the [IFC](https://en.wikipedia.org/wiki/Industry_Foundation_Classes) file format.

The BCF format implements a series of tools to allow users to create, annotate, classify, comment, discuss and solve issues on a BIM model. It works as a thin layer containing issues related to a BIM model. The BCF file itself doesn't contain or relate explicitly to the model, which allow different people who work on a same project but with different applications to still be able to discuss and process the same issues.

This project idea proposes to implement a BCF viewer for FreeCAD. The viewer would allow to:

* Open BCF files
* View the different issues with their comments and images (screenshots)
* Navigate in a FreeCAD model to a camera position stored in the BCF file
* Comment, annotate, classify and tag issues
* Create new issues, save screenshots and camera position
* Save BCF files
* If time permits, explore the possibilities of server-based BCF serving (API)

## Details

* Get familiar with FreeCAD and the current state of BIM tools
* Get familiar with [https://en.wikipedia.org/wiki/PySide PySide] and Qt
* Get familiar with the BCF format, gather a series of test files
* Implement the different functions needed to interact with BCF files and data
* Implement a Graphical User Interface to these functions

## Expected Outcome

* Documentation. Like all open-source projects, it is of uttermost importance to document a lot everything you do. This allows for other people to take on and extend your work more easily.
* A complete, working GUI application, that works as a plugin/addon inside FreeCAD.

## Future Possibilities

BCF use in the BIM industry is still at its infancy. One can think of many possible extensions, such as:

* Connecting with other (open-source) communication structures: forums, chat/messaging systems, etc...
* Connecting with other issues tracking systems

## Project Properties

### Skills

* Programming language: Python + PySide
* Knowledge of BIM modeling and workflows is a plus, or be ready to practice!

### Difficulty

Medium

### Additional Information

* Potential mentors: [ Yorik](http://forum.freecadweb.org/memberlist.php?mode=viewprofile&u=68)
* Link to this idea on the FreeCAD wiki: https://www.freecadweb.org/wiki/BCF_support

Vanuan, 2019-08-02:
New link to BCF https://technical.buildingsmart.org/standards/bcf/

yorikvanhavre, 2020-02-10:
This has been done in 2019 so I'm closing it.
The result is available at https://github.com/podestplatz/BCF-Plugin-FreeCAD

closed 2020-02-10 by yorikvanhavre