opencax/GSoC #17: Direct Modeling tools

yorikvanhavre, 2019-02-26:
## Outline

[FreeCAD](https://www.freecadweb.org) is a feature-based parametric modeling system, hence different modeling steps depend on one or multiple previous steps. Each step is at the same time a tool/operation, and a geometrical object, resulting from that operation. This mix of operation and object is called a feature. Direct modeling, often presented as the opposite of parametric modeling, allows to graphically move vertices, push or pull faces and edges to modify the geometry of an object.

Watch an introduction to direct modeling on https://www.youtube.com/watch?v=RMrvjAPxyN0

The two concepts could however easily coexist in FreeCAD. Each operation performed via direct modeling could be stored into a new feature, that would successively modify the previous object by adding transformations, such as a displacement of a vertex, an edge, or a face, or extrusion of a certain face.

This GSoC project aims at throwing the bases of the necessary feature(s) to hold such transformations as parameters, and the 3D manipulators to allow the user to modify an existing 3D object by clicking and dragging on the screen.

## Details

* Get familiar with OpenCasCade, FreeCADs geometric modeling kernel, and understand the geometry structure, and how complex 3D objects are constructed.
* Get familiar with Coin, FreeCADs 3D display manager, and understand how to create manipulators and interact with the 3D scene.
* Understand how FreeCADs parametric objects (features) are constructed and work
* Study and document existing direct modeling applications and techniques through available online videos and identify potentially useful direct modeling operations to be performed in FreeCAD. Study which pieces of information each of these operations requires, and how to store that information using FreeCAD's available parameters.
* Throw the bases of a feature, or a family of features, that can hold the necessary information, and perform the transformations. This can be done in either C++ or Python, or, ideally, a C++ core extensible in Python.
* Throw the bases of a manipulator, or family of manipulators, to allow the user to perform transformations by pushing and dragging on-screen, and experiment with ergonomy and usability. This can be done in either C++ or Python.

## Expected Outcome

* A documentation of the various possible and useful direct modeling operations that should be performable in FreeCAD, and how each of them would translate into parametric objects (features).
* An implementation of a base feature, or family of features in FreeCAD, that can perform basic operations (proof of concept), that can be easily extended to support more operations in the future
* Experiments and/or implementation of a Coin manipulation system, that would provide the FreeCAD user with nice and ergonomic ways to interact with the geometry and perform transformations. This could also benefit other areas of FreeCAD.

## Future Possibilities

Since direct modeling is a huge area in constant evolution, the work done in this GSoC will only cover a small part of it. But if the bases are done right, extending them will be easy and could develop very far.

## Project Properties

### Skills

* Programming language C++ or Python
* Good understanding and use of APIs from FreeCAD and OpenCasCade 
* Knowledge of 3D modeling, topology and computational geometry is a plus

### Difficulty

Medium

### Additional Information

* Potential mentors: [ Yorik](http://forum.freecadweb.org/memberlist.php?mode=viewprofile&u=68)
* Link to this page on the FreeCAD wiki: https://www.freecadweb.org/wiki/Direct_modeling_tools

TheBeLieveRs, 2019-11-11:
I am Prakhar Srivastava a student if IIT Roorkee ,India and I would like to work on this issue can u tell what ide u suggest for this, or maybe the ide u would have worked on

yorikvanhavre, 2021-01-29:
**Warning**: This project idea had been formulated for GSO2020, and might be too big for the new format of GSOC2021. Nevertheless, this functionality is still required, and this can serve as a base to build your own project, or could be implemented partially.

yorikvanhavre, 2021-03-03:
Other FreeCAD ideas for 2021 at https://www.forum.freecadweb.org/viewtopic.php?f=8&t=54913

MariwanJ, 2021-03-09:
@Yorik: 
By parametric modeling : Do you mean that even Direct Modeling should have options, parameters that could be changed live to change the object you applied a command before? 
Like what you have now in Sketcher? 
In my understanding, direct modeling will not have any parametric, you apply a command and the object will follow that command .. that is it. 
You might have some undo-redo but not more. 
If you think to use parameter to change an object, that would create a huge problem for the project. It is like class-inheritance. if someone, or somehow the base-class changed .. the whole project might fail. 
Please forgive my lack of knowledge. In all commands I am implementing, I make a simple copy of what I done. So no way to go back a step. I didn't implement yet undo-redo .. but there will be .. More than that will not be possible. 

Biggest problem in FreeCAD at the moment for me .. Many object-parameters are read only. So you cannot easily modify them. 
For example chaining location of a vertex will not be possible if you don't re-crate the object from new vertices . you need to recreate them to be able to change anything. 
That is what the API tell me .. I might be wrong.
First thing to do in this project. Make tools (classes) to let developer use them. Tools that allow developer to modify shapes, vertices and objects. Documented in details. 
Second thing to mention: 
Should we stick to CPP or it is ok with python. I have a CPP background but started to use python since it was easier to start with FreeCAD API.
Unfortunately, There are so many developers out there that make a different functionalities, but but it is totally fragmented .

yorikvanhavre, 2021-03-10:
You approach seems good too, it's mostly a question of deciding how one wants to do things. There is basically no right or wrong here I think. Indeed tying strictly to parametric model has drawbacks too (slower approach, maybe ends up in a "half-baked direct modeling"), But the current API can be extended too. Having "movable vertices" is possible too, look at the Draft Wire/polyline for example. It's just a matter of how you design things.

MariwanJ, 2021-03-10:
Yorik: 
You are an experience FreeCAD developer. Please answer my question regarding if I stick to CPP or Python?
I have a feeling that it would be difficult to make all these bases/changes using python .. at least it will be slow somehow.

luzpaz, 2021-03-10:
If you use `cpp` and want people to test, then we'll need to figure out how to compile FC test builds for people to use. Most won't know how or not want to because of the amount of time it takes to compile FC.

yorikvanhavre, 2021-03-10:
> Please answer my question regarding if I stick to CPP or Python?
Basicallyit does not make big difference... It it more a question of your own preference, what you are most comfortable with. Almost anything can be done in either language. If you are thinking of doing this in a separate addon/workbench, though, then working in python is an advantage, as your code can be distributed "as is", no compiling/packaging necessary.

MariwanJ, 2021-03-13:
Yorik. Do you think that We need to start making a new branch of new base classes where we driver the shapes from the TopoShapes API?.
Or how should we start? At the moment I am trying to learn by adding some small commands, new 2D/3D objects to my WB. But that is not the way to achieve this project IMHO.
Please would you write in some sentence how this goal should be achieved in your opinion?. How could we have 3D primitive object that is easy to do all kind of modifications? (Extrude, Cut, Shell, ..etc) . The classes itself should provide the tools. Not by creating tools that re-create the object.

yorikvanhavre, 2021-03-16:
In my opinion there is no need to touch the TopoShapes themselves. You would be reinventing the wheel, which is already well enough done by OpenCasCade. 

I think this is mostly an UX job, not a geometry job. What we need is to think a system of how a person can interact with a shape (ex: move a vertex, move an edge, move a face, scale an edge, scale a face, rotate an edge, rotate a face, extrude/intrude a face, extrude several faces, move several vertices/edges/faces, cut through/divide a face, the list can be endless), and define UI tools that provide a nice UX experience to the user to perform these things. That is, define the steps of what the user does, what they see, what they can do next, etc. and implement the needed coin workflow (manipulable widgets, etc) That's 99% of the work. Recreating the object itself is 1% of the work (OK maybe I'm exaggerating a bit ;) ), and IMHO it's no big deal to have the tool recreating the object, and it's overkill to re-imagine a whole geometry system just for that, when we have a fairly good one already. 

But of course all this is just my vision, basically it is how I would myself approach it. Any research is always a benefit, so pursuing another way is absolutely no wasted effort, if you are interested in doing it that way.

MariwanJ, 2021-03-16:
Let me explain how I look at the matter: 
Early days when I started learning CPP, I started using FLTK. It is the simplest and best cpp project that teach you every expect how windows drawing works, classes, inheritance ..etc IMOH. I will try to use the same angle of view.
Each object consist of vertices as base class.  Vertices will get its draw() function which draw a point on the screen. 
There will be a move function as a virtual function that children to Vertices will implement their own (like line, square ..etc). 
When you come to 3D shapes, it should be constructed as a list of shapes. Their Draw function should take care of drawing the object from the collection of the children. 
i.e. no object is a complete object .. it is a collection of small father classes ( face or lines or vertices). 
Since you have control over each vertices in the base classes .. it wouldn't be a big issue to move, extend ..etc.
To draw a face, you need to go again to the API provided by the TopoShapes and their API. 
For some reason there will be some object that you cannot using TopoShapes. At that time the classes themselves should be able to draw that. 
There must be away to join, combine, delete, cut ..etc virtual functions in each class and their inheritance. 
That is how I look to the project. TopoShapes made it impossible to use their API to recreate easily whatever shape you want to manipulate. So you need to construct them (draw them). 
Let me know what you think about this idea .. and how different is compared to your view.

yorikvanhavre, 2022-02-08:
This is still valid for GSOC 2022. One should have a look at other attempts and maybe joint forces:

* carlopav in https://forum.freecadweb.org/viewtopic.php?t=49837
* Design456 workbench in https://github.com/MariwanJ/Design456

KeithSloan, 2022-03-16:
"What we need is to think a system of how a person can interact with a shape (ex: move a vertex, move an edge, move a face, scale an edge, scale a face, rotate an edge, rotate a face, extrude/intrude a face, extrude several faces, move several vertices/edges/faces, cut through/divide a face, the list can be endless)"

Given that this project is defined as Hard and the above long list of interactions, how about starting with a sub project namely interacting with Part2DObjects [https://wiki.freecadweb.org/Part_Part2DObject](url)

Once the foundations of interacting with 2D objects is established then things can then progress in the fullness of time onto the full gambit of Shapes and the multitude of possible interactions

i.e Initial phase add support in FreeCAD for 

- Selecting

      * The ends of lines
      * lines / curves
      * Control points
      
- Moving selection
- Updating Part2Object info

Then could update the Curves part of the Curves workbench to have editable curves, this could be done without the need to recompile FC.

A follow on step could be Surfaces following a similar approach.

I feel the above would make for a more containable project with a useful end point

MariwanJ, 2022-03-17:
@KeithSloan 
This is not anything bad for what you wrote but or against anything/one, but it is the reality of FreeCAD. This is MHO. 
It is easy to describe what we should do for making direct modeling achieved. But the reality is the OCC is not capable to do so. It is very slow, and full of bugs. 
I made several tools and those tools are suffering from the slowness of the OCC and FreeCAD. 
Several times I described how we can try to introduce other things to FreeCAD but the freecad community are not interested in direct modeling. 
Along, I was trying to talk the forum for finding other way of thinking.. the result is nothing. 
So, it is not possible to make what you described simply because freecad cannot do direct change in objects due to the internal structure for both freecad and OCC. 

If you read about 3D Wings you understand the principle used there to achieve the direct modeling they have. 

For every little change you do in freecad,  you have to redraw the whole thing. In normal Windows/X server GUI , often you have to limit the area you are redrawing/recalculating. That is not possible for OCC or FreeCAD. 
I have a test that I just collect some 3D models in a Fusion object. It becomes so slow after some 30 pcs that is not responsive any more as it should... only a Fusion nothing complicated. I have a video about that.

Without real change in the way FreeCAD made, real direct modeling is no just hard, it is impossible. 
Look at and use my tools and my videos, you will see the difficulties.
It is almost a year I am trying to work on this project, still I am along .. no one wish to come to the project. 
hope I answered your comment. 
Google summer will results in nothing to be honest.

KeithSloan, 2022-03-17:
@MariwanJ I have in the past had a brief look at your Design456 Workbench. I think it is brave attempt at Direct modelling but I am not sure your about your approach of trying to enable manipulation via OCC.

@yorikvanhavre Makes second item in details is "Get familiar with Coin, FreeCADs 3D display manager, and understand how to create manipulators and interact with the 3D scene" i.e  Looking at implementing Coin facilities to enable manipulation.

Some time ago I added an issue to the Curves workbench see[https://github.com/tomate44/CurvesWB/issues/59](url)
asking for the ability to edit curves the response "Making the interactive feature is not an easy task with Pivy.
@joelgraff made the [pivy_trackers](https://github.com/joelgraff/pivy_trackers) library that looks great. I did a couple of tests but was not able to use it properly, unfortunately."

I think Joels tracker is used in the Civil Engineering workbench Trails perhaps @joelgraff and @HakanSeven12 could comment.

@MariwanJ "It is almost a year I am trying to work on this project, still I am alone .. no one wish to come to the project.
hope I answered your comment. Google summer will results in nothing to be honest."

Which re-enforces my view that this needs to be tackled as a small subset, my suggestion being just manipulation Part2DObjects. 

SMALL STEPS and SMALL BEGINNINGS

MariwanJ, 2022-03-17:
Regarding coin, I made many detailed steps how to use it and I have the widget system FR_WIDGET. But I don't like the idea to use coin. That makes my code too complex with a gain of nothing. You wouldn't be able to replace OCC as you need to redraw the shape each time you change the (face, vertex, edge) .. just not possible. For very basic shape, I managed to replace OCC with coin .. but that is far from being used properly .. and works only on planer shapes.
 
Newer 3d library is essential and a better way of representing objects in 3D world is also a MUST. 
I described my idea at the forum .. no one replied that post and for that reason .. there is no way to do that thing by a person. 
You have to take that idea.. you shouldn't redraw the whole screen because of a change that is made on one object. If you are careful in looking at other CAD program (for example DesignSpark Mechanical) you will notice that internally they keep the (cylinder, box, ..etc) as it is .. but they draw the results not the primitive objects. As soon as you click the object, they draw the basic shape behind that part or face. 
Any external library will fail since FreeCAD will "never" included and your project will not reach users. That is why I am kinda rejecting including other library without been added to FreeCAD core.

KeithSloan, 2022-03-18:
Question - With a Sketch I can create a BSpline and I can edit it by pressing the Ctrl key and selecting a Control point which I can drag around and the curve changes in realtime. Okay this is 2D but how is the Sketcher workbench implementing this?

MariwanJ, 2022-03-18:
Each time, it must recreate the whole thing. There is no other way to do so. Think if you have hundred of objects, and OCC is slow? 
it wouldn't work.

KeithSloan, 2022-03-18:
Well it does not appear to redraw the other lines in the sketch, just the BSpline

luzpaz, 2022-03-18:
And aside,@realthunder was talking about Coin3D in foss4g. I don't recall if it was for replacing or augmenting the coin3d library? Perhaps he can weigh-in?

realthunder, 2022-03-19:
It is to augment Coin3D. I'd like to keep the node construction and iteration part of Coin3D, but replace its rendering backend with some third party library. The current design of Coin3D rendering backend is too tightly coupled with the rest. I managed to use its SoCallback interface to replace its render caching entirely and perform rendering with my own code. Although the rendering is done using only OpenGL 2 which is more or less the same as Coin3D, there is already big performance gain for large assembly.

The most critical part for improving the rendering performance is actually the rendering cache. In CAD particularly, where the assets are dynamically created and prone to changes. The cache must be able to reuse as much as possible but can still perform incremental update efficiently. Using third part library makes it easy to leverage on modern graphics API. Apart from giving us more eye candy, ultimately, for a CAD system, it is to help improve caching, like instance rendering, etc.

OCC boolean operation is no doubt slow in non trivial cases, but we can work around it by showing partial result. Like what I've done with PartDesign right now, for almost all features, you can preview only the tool shape while editing instead of the final shape. We can even explore possibility to run OCC and rendering in different CPU cores.

BTW, I am very much interested in direct modeling. But I think the rendering part is more important at the moment, which is what I'd like to work on first. Actually, there are two not quite related aspect of direct modeling, one is the push pull UI, and the other is the much more difficult feature-less/history-less direct shape manipulation like those in Fusion 360 and SolidEdge. I'm assuming we are talking about the first aspect right now, although I'm interested in both.

aothms, 2022-03-19:
I'm not really involved in FreeCAD, but I feel the urge to comment as well:

Even if it turns out direct modelling on top of OCCT is slow for interactive use. There might be cases where it might still make sense in programmatic use. For example. I'd like to have the option to snap the second operand geometry onto the geometry of the first operand prior to performing a boolean operation (giving more local control, as an alternative to the global fuzziness). I can imagine that out of this project a good API originates that could be used for that purpose. Like `ShapeBuild_ReShape` but with a more geometric focus.

Yes, OCCT can be slow, but it might be that certain bottlenecks can be eliminated with caching or a more efficient higher level data model (Maybe Part already provides some of that, sorry I don't know). For example, for calls like `OuterWire` one might naively assume it's like a constant time operation that just reads a flag or something. Where in fact it does a full geometric check for containment of a point infinitely far away for every wire in the face. If the bottlenecks are known I think that with an intermediate data structure on top of TopoDS it may be possible to provide a more efficient and more user friendly API for this purpose. Same for ancestor and sibling lookup, they can probably be stored directly and with the assumption of a manifold valid data structure certain shortcuts can be made in terms of retrieval.

MariwanJ, 2022-03-19:
Let me explain how I think about the FreeCAD structure. It would be beneficial if you look at other CAD to know how they define the basic elements (Vertex, line, face) like [Wings3D](http://www.wings3d.com/ ) .  
For freecad to gain a real improvement, we need to think totally different. We need to think in the basic elements not solid objects. Each basic element should have it's own redraw mechanism. Which by that you can decide when you need to redraw. 
It is not essential for the user to see a solid object always rather than a drawing represent the solid object. For export and save-as you generate only once the solid object. 
Now, knowing that, the mechanism should be enough cleaver to eliminate basic shapes whenever they be inside other objects, or cut, join them and produce a new object based on the old object. Since they are basic elements it wouldn't be too difficult.
These are only data structures, not a drawing .. it will be fast. Redrawing is done only when the basic element get a damaged signal and you call a redraw to that part only when it is visible and you are trying to show the final result to the user. 

Simplifying the thinking like that will help in eliminating the need of OCC in the scale FreeCAD uses at the moment. 
Let all shapes be based on a collection of those three basic elements.
Coming to direct modeling: Whatever you do to overcome the OCC problem will result in an ugly code, ugly performance and in the end useless tools. 
Restructuring FreeCAD is the only way to make freecad reach the level of other CADs. With modern graphical libraries, and with a view of having a bigger goal than temporally fixing some problem we see now. This will break back compatibility but that should be done. Maybe tools must be made to convert the format later.
FreeCAD developers must agree on these proposed changes of mine to go further with the development of direct modeling. I see no future for all repair attempt people discuss and try. We have to start from scratch and think different. Take benefit of what other made. 
Rather than simulating the faces and edges. They should be the real part in the drawing not a COIN3D part. Double as much you are drawing and double as much you are calculating now. Not difficult to understand why FreeCAD is so slow
Note:
My ideas are taken from how normal GUI works (take an example of the good structured FLTK ).

MariwanJ, 2022-03-20:
A real profiling of the code might find other weak points in the whole project. Don't know if anyone did that but I don't think OCC is the only weak point. Might be other weak points which we don't know. Visual profiling of the whole code will tell us where time are consumed. This is something that the core developers needs to do.

MariwanJ, 2022-03-22:
@realthunder 
I would like to ask you what if you ever thought about add new 3D graphic library (modern) which might last long time and give us more capabilities? 
This is just an example (https://www.ogre3d.org/) 
What are the steps one should take to add any new 3D library to replace COIN3D?
I would appreciate any tips, or how to do?
Any one else can answer this question is welcomed. thanks in advance

brlcad, 2023-02-23:
FreeCAD is now participating as their own GSoC org, congrats!

closed 2023-02-23 by brlcad

MariwanJ, 2023-02-23:
@KeithSloan 
COIN3D is a dead project, OCCT is kind a closed way (no fix is expected).. so no hopes there at least for now. I tried to find a way to inject OpenGL (modern) into COIN3D, but no .. it is not possible. 
The only thing remained for me is try to bring Wings3D to freecad. 
But the problem is Wings3D is written using a such horrible language that you cannot bring it so easy. And also is based on very old OpenGL. 
I tried to use FLTK, it has also a lot problem. Wasn't easy. 
I know try to make the conversion using ImGui. 
Nothing is easy.  Design456 based on OCCT and COIN3D is almost dead because of all difficulties I faced for direct modeling. 
FreeCAD because of OCCT is veryyyyy sloooooow. And direct modeling is about quickly movements .. so that is not a good combination. 
in short, no direct modeling is possible without using MESH and using OpenGL (modern) or Vulkan.

MariwanJ, 2024-03-10:
> Hey @MariwanJ, just came across this comment.
Hi, Thanks. 
I explained at different occasions what the problem is with FreeCAD for direct modeling.  It is the engine. 
I am not sure if COIN3D is so fast that can be used in mesh manipulations. What I need is to have a working, multi-threaded mesh manipulator platform. 
That is what I am trying to achieve with Design456App. 
Not sure if Coin3D can do that. It is also complicated for no reason. 
Another thing that I am not sure how to achieve is targeting mesh structure. I mean finding edges, faces ..etc based on selection. Cut, join, ..etc of faces, mesh elements .. all that stuff. 
Basically, you need a good, multi-threading manipulator of mesh to achieve direct modeling on mesh not OCCT.
Without working on mesh, direct modeling is not possible for FreeCAD. 
Not sure how you can modernize COIN3D. seems to be a lot of work.
BR

yorikvanhavre, 2024-03-11:
This issue is now happening on the FreeCAD repo at https://github.com/FreeCAD/FreeCAD/issues/5764
Please continue the discussion there instead!