opencax/GSoC #23: LibreCAD GUI revamp

feragon, 2019-02-27:
## Outline
LibreCAD 3 UI is written in Lua and Qt, using a custom file to bind the Qt methods to Lua.

This project consists of rewriting the GUI in C++.

## Details
Every [GUI file](https://github.com/LibreCAD/LibreCAD_3/tree/master/lcUILua/ui) should be rewritten in C++, and keep the operations written in Lua.

That implies to remove the Qt signals connections to Lua slots, and call Lua functions in C++ slots. 
Direct Qt widget creation will be impossible after, so an API should be accessible to Lua plugins to create elements in the interface. 

The interface can be restyled too, and its integration with the system improved.

## Expected Outcome
Remove the [Qt bindings](https://github.com/LibreCAD/LibreCAD_3/blob/master/lcUI/lua/qtbridge.cpp) to Lua.

Unit tests are expected.

## Project Properties

### Skills
- C++
- Lua
- Qt

### Difficulty
medium

### Additional Information
* Potential mentor(s): [Florian](https://github.com/feragon)
* Organsation website: https://librecad.org/
* Communication channels: https://librecad.zulipchat.com/

Jedi18, 2019-12-23:
Hello, I would like to know if the issue has been resolved. I am interested in working on it.

lordofbikes, 2019-12-24:
This is not an issue, it's a rough project idea for Google summer of code.
The idea is, that a GSoC student make a proposal based on this idea, how he would work on this task during the GSoC period.
Anyhow, nobody has worked on this now. So if you are interested in working on it, I recommend to get in touch on our Zulip channel, which is linked above.

closed 2021-01-29 by lordofbikes