opencax/GSoC #26: Online Geometry Viewer (OGV)

brlcad, 2019-02-28:
## Outline

Write a proposal that leverages or otherwise improves OGV!  This can be backend (C/C++) or front-end (Javascript) work, but should focus on completeness for an end-user.

## Details

We have been working on OGV for over 5 years. It started with PHP and then was revamped to meteor.js (before it was cool).  We want to focus on the backend of OGV, making sure it works properly, converts the models properly, and basically finish a 1.0 version of OGV so we can launch it for the masses!

We have arranged a list of features, bugs, and enhancements that we think should be handled this year and have put them (https://github.com/BRL-CAD/OGV-meteor/projects/1?fullscreen=true) here on github. These issues are arranged via priority, so you will get an idea about what you can think about including in your proposal for this summer.  You don't have to limit yourself to those ideas.

Checklist to write proposal for OGV
* Download and clone OGV from https://github.com/BRL-CAD/OGV-meteor
* Setup and Run OGV on your local machine.
* Fork OGV repo
* Choose a couple of low-hanging issues, solve, and send a pull request.
* Talk to mentors about your PR
* Choose list of issues that you would like to solve this summer
* Make a detailed weekly implementation plan
* Share your proposal with your mentors
* Submit it to the GSoC website

## Expected Outcome

You're expected to propose an outcome **useful** to end users.  That is a broad range of possibilities that will depend on your interests and experience level.  For example, you might propose focusing on the backend conversion to triangles for display (C/C++).  Or you might propose changing the backend to NURBS surfaces (C/C++) and using [verbnurb](http://verbnurbs.com) or three.js (Javascript) to display them instead of triangles.  Or you might propose keeping backend the way it is and focusing on front-end robustness (Javascript), website features, or deployment infrastructure.  You hopefully get the idea.

## Project Properties

### Skills

* Javascript
* C/C++ (optional)
* Verbnurb (optional)
* Three.js (optional)

### Difficulty 

medium

### Additional Information

* Potential mentor(s):
[Shubham Rathore](mailto:shubhamrathore1947@gmail.com) 
[Harmanpreet Singh](mailto:harman052@gmail.com)
[Gauravjeet Singh](mailto:gaurav.ishwerdas@gmail.com)
[Inderpreet Singh](mailto:indrplus@gmail.com)
* Organization website: https://brlcad.org
* Communication channels: https://brlcad.zulipchat.com

prakasharyendra, 2020-03-13:
Interested to contribute.

mahmodHammad, 2021-03-19:
Interested to contribute

brlcad, 2021-03-19:
You should join us on our Zulip chat to discuss.  Welcome!

divyanshugarg36, 2022-03-09:
Interested to contribute.

brlcad, 2022-03-09:
Interest is always welcome.  Please join us on https://brlcad.zulipchat.com to discuss.

divyanshugarg36, 2022-03-09:
Yes, I am already there, I checked this project and I am currently learning the working of this project.
Found out that we are planning to migrate it to the latest technologies and I think I have a lot of interest in these techs mentioned in the specifications.

DrSensor, 2025-02-20:
I try to build the Dockerfile in the OGV-meteor repo but got this error:
```
STEP 4/6: RUN mkdir build && cd /app && meteor build /build && cd /build && tar -xzf app.tar.gz && rm -rf app.tar.gz && cd /build/bundle/programs/server && npm install && rm -rf ~/.meteor
Unable to update package catalog (are you offline?)

If you are using Meteor behind a proxy, set HTTP_PROXY and HTTPS_PROXY
environment variables or see this page for more details:
https://github.com/meteor/meteor/wiki/Using-Meteor-behind-a-proxy 

This project says that it uses Meteor 2.2, but you don't have that version of
Meteor installed, and we were unable to contact Meteor's update servers to find
out about it. Please edit the .meteor/release file in the project and change it
to a valid Meteor release, or go online.
Error: building at STEP "RUN mkdir build && cd /app && meteor build /build && cd /build && tar -xzf app.tar.gz && rm -rf app.tar.gz && cd /build/bundle/programs/server && npm install && rm -rf ~/.meteor": while running runtime: exit status 1
```
seems like they deprecate their update server 🤔

do you have the pre-build docker image for this?

divyanshugarg36, 2025-02-20:
> I try to build the Dockerfile in the OGV-meteor repo but got this error:
>
> ```
> STEP 4/6: RUN mkdir build && cd /app && meteor build /build && cd /build && tar -xzf app.tar.gz && rm -rf app.tar.gz && [...]

Hey @DrSensor,
Unfortunately, we don't have a docker pre-build, but I would recommend you to go with the non-docker way which worked for me 1 year ago, also I believe you can manually install the meteor version using https://www.npmjs.com/package/meteor?activeTab=versions
Or better to experiment, install Meteor using the NPM registry and try out docker.