opencax/GSoC #27: New BRL-CAD GUI

brlcad, 2019-02-28:
## Outline

Implement the beginnings of a new GUI for BRL-CAD!

## Details

BRL-CAD has two main graphical applications called 'mged' and 'archer' which look like they were developed in the 80's and 90's respectively (because they were).  We need a modern GUI, ideally something based on a modern framework like Qt (preferred) or Electron (convince us) or something else (convince us).

This new GUI will need to leverage our existing libraries in a big way, LIBGED in particular which is basically all the command-line commands available to both mged and archer.  You'll need to propose some way for displaying geometry, e.g. wireframes or triangles via OpenGL or real-time ray tracing via Intel OSPRay+LIBRT.  It must work on BRL-CAD's .g files.

## Expected Outcome

A minimally functionally complete new GUI!

## Project Properties

### Skills

* C/C++
* depends on framework

### Difficulty 

medium

### Additional Information

* Potential mentor(s): [Sean Morrison](mailto:devs@brlcad.org)
* Organization website: https://brlcad.org
* Communication channels: https://brlcad.zulipchat.com

math-alpha, 2019-03-30:
I am interested on working on this. As soon as i send a PR, i will notify on this thread.

prakasharyendra, 2020-02-07:
interested, from where should I start?

brlcad, 2020-02-09:
> interested, from where should I start?

@prakasharyendra unlimited possibilities, but I would suggest doing the mged tutorials (which you can find on the main website) so you understand basic current usage, and then work on some minor patch (see BUGS or TODO files in an svn checkout).

ravenhaer, 2022-03-05:
I am interested in working on this. Should I try to solve some bugs/to-do before trying to create a mock-up of the UI? What are some of the major UI changes/features we are aiming for?

brlcad, 2022-03-05:
I would definitely recommend researching the links and existing efforts that are included in the idea description.  Having a mock-up on the UI is (probably) not as important at this point as having a clear understanding of what the current UI looks like, what the existing arbalest GUI looks like, what the current qged GUI looks like, as well as the current production tools (mged and archer), so that you can propose something meaningful.  A patch that solves a specific bug/to-do would be a great next step along with discussing your ideas with our folks on Zulip chat.

Mr-codehunter, 2023-03-13:
I'm interested in this project and I want to participate in this project.

I have done some work on Qt framework. it is a c++ framework.

Mr-codehunter, 2023-03-13:
@mentor Can you give me a starting point to work on this.

Mr-codehunter, 2023-03-14:
> You already joined our Zulip channel, right?

Yes sir.

XaXtric7, 2025-03-01:
@drossberg Hi , i am new in this community and the whole GSOC thing and i really want to contribute.

drossberg, 2025-03-02:
Hi @XaXtric7, thanks for your interest in contributing to BRL-CAD.

You should start with familiarizing yourself with BRL-CAD. I.e., install the programs and go through the tutorial. Then, you should join our community chat https://brlcad.zulipchat.com. This is the place to ask questions, e.g. about building the programs by yourself, which is a prerequisite for contributing code.