opencax/GSoC #28: Implement support for other annotation styles

brlcad, 2019-02-28:
## Outline

Preliminary support for annotations was added in [2017](http://brlcad.org/wiki/Google_Summer_of_Code/2017#Annotations_for_the_wireframe_view).  Take that support to the next level!
 
## Details

Annotations are labels, dimensions, tolerances, and other notes that you can specify on geometry or they can be simple notes that are displayed. 

BRL-CAD already has an "annotation primitive" that holds basic annotation information (labels) but it's very basic and not yet usable.  Your proposal should explain in detail what annotations you intend to prioritize first and how you'll go about adding that support.

## Expected Outcome

The end result should hopefully be something like annotation examples you can find on the web (e.g., https://www.3dcadworld.com/leveraging-3d-cad-data/ ).
 
## Future Possibilities

There's not enough time to implement a complete GD&T system during GSoC so be sure to plan to implement, test, and demonstrate one feature at a time in your proposal.

## Project Properties

### Skills

* C/C++
* Vector math

### Difficulty 

easy

### Additional Information

* Potential mentor(s): [Shubham Rathore](mailto:devs@brlcad.org)
* Organization website: https://brlcad.org
* Communication channels: https://brlcad.zulipchat.com

AliHaider93, 2019-03-28:
hello , my name is Ali Haydar and  I want to participate in this project is there any advice to me . I have read the work in the  2017 and I am not very familiar with the namespace and the header files like magic.h
and after we define the geometry of the annotations is there a specific way to draw the result on the mged graphic window ?

brlcad, 2019-03-28:
Hello Ali!  After you define an annotation, you simply "draw" it and it should display in the mged graphics window.  I suggest going through a couple of the MGED tutorials so you understand basic usage, but you can also join our Zulip chat for interactive discussion.  As for advice, I suggest you do the MGED tutorials, try to make an annotation ( read http://brlcad.org/wiki/Annotations and http://brlcad.org/wiki/Annot ), and then figure out what to propose from there.  We want to start simple, so don't worry about trying to understand everything.  You won't have time for that. :)

aamitn, 2021-03-27:
Hello, really looking forward to contribute to BRL-CAD on the Annotation Implementation part as a part of contribution towards gsoc 2021

Hope im not too late

brlcad, 2021-03-27:
Hello @aamitn !  You’re not too late but you will have to do a lot of work quickly to demonstrate your ability to read code.  Please join us on Zulip and work on a patch as quickly as possible.

niralverma, 2021-03-28:
Hello, I joined today. Can I know from where to start? I am interested in this project.

niralverma, 2021-03-28:
Hello. I joined today, I know its really late. But I am interested in this project. Can I please know from where to start? 
And if there is any hope for me at this last moment ?

Himanshu40, 2021-10-18:
Hello, I am really looking to work on this issue. I went through the work that already had been done in 2019 by AliHader. May I have to dig some more? Need guidance. Thanks.

Himanshu40, 2021-11-04:
> Hello Himanshu40, If you have any questions I am available to answer. Ali

Hello @Alimaff. Can you help me where to look at the codebase to get familiar with? I learned how to use mged application and also covered some topics in mged tutorial.

closed 2023-02-23 by brlcad