opencax/GSoC #32: NURBS Editing Support

drossberg, 2021-01-28:
## Outline
Implement the prerequisites for NURBS editing in BRL-CAD's GUIs.

## Details
BRL-CAD has support for raytracing of NURBS surfaces implemented, but they are handed over as BLOBS to the openNURBS library.  Beyond basic operations such as rotation and translation, the BRL-CAD core has no ability to edit them.  This project would implement support for editing NURBS curves and surfaces in the BRL-CAD core, thus creating the prerequisites to handle them with higher level (i.e. GUI) tools.

## Expected Outcome

## Future Possibilities 

## Project Properties

### Skills

### Difficulty 

### Additional Information

* Potential mentor(s): [NAME](https://)
* Organization website: https://
* Communication channels: https://******.zulipchat.com

Forchapeatl, 2025-03-29:
Hello @drossberg  and @brlcad  . Please I would like to work on this issue for GSOC 2025. I will start with the good first issues and gradually build my way up to this Level. Thank you.

drossberg, 2025-03-29:
Hi @Forchapeatl, thanks for your interest in BRL-CAD.

You should start with learning about BRL-CAD and how to build it. You can do this in any order: Install BRL-CAD from a [binary release](https://github.com/BRL-CAD/brlcad/releases), do the [tutorial](https://brlcad.org/w/images/c/cf/Introduction_to_MGED.pdf), and build the programs from sources afterwards. Or, you start with building and installing the programs, and do the tutorial then.

For discussions, we have our [Zulip chat](https://brlcad.zulipchat.com). Please, do the further conversation there.