opencax/GSoC #33: Font Selector Window with preview and search functionality

t-paul, 2021-02-13:
## Outline

OpenSCAD supports extracting geometry of text based on fonts installed on the system. The existing font list gives an overview of what fonts are installed, but does not provide further help to select fonts. An improved selection window could improve the user experience by giving the user a preview of some sample text and more ways for selecting a font.

## Details

* The current font list is implemented as dialog, this should be changed to a dockable window similar to the console
* Preview area should allow the user to enter some (limited) text which is then rendered in either a selectable size or maybe in multiple sizes
* Allow font selection by character, this will allow the user to find fonts that support specific unicode codepoints (e.g. for emoji)
* Select by font name + drag & drop to the editor - keep the feature of the existing font dialog

## Expected Outcome

The font selection dialog is replaced with a dockable font selection window with extra functionality.

## Project Properties

### Skills

* Programming language is C++
* GUI programming with the Qt framework

### Difficulty

Medium

### Additional Information

* Potential mentor(s): [Marius Kintel](https://github.com/kintel) ([IRC](https://webchat.freenode.net/?channels=openscad): kintel), [Torsten Paul](https://github.com/t-paul/) ([IRC](https://webchat.freenode.net/?channels=openscad): teepee)
* Organization website: https://www,openscad.org/

sky3760000, 2021-03-24:
Hi ,
I'm interested in this project ,please guide me through.

t-paul, 2021-03-24:
@sky3760000 please see [Getting Started](https://github.com/openscad/openscad/wiki/Google-Summer-of-Code---Getting-Started)

aykalkan, 2021-04-01:
I've built OpenSCAD from source files. I'm trying to increase my familiarity with it. Where exactly font list appear when we try to extrude a text? There is only a font parameter given to a text object as I see. I can only preview font of the text when I hit F5.

t-paul, 2021-04-01:
Correct, that's one of the things this change would make possible - preview how a font looks in the GUI without having to use `text()` in a script and execute that script.
So the focus is on GUI only, changing the dialog that is displayed with `Help`->`Font List` menu entry to dockable window like the Console, or Error Log. This window should then have the additional features listed in the project description in addition to what the existing dialog does.

closed 2022-02-07 by t-paul