opencax/GSoC #34: Integrated language help feature

t-paul, 2021-02-13:
## Outline

Add more interactive help features for built-in functions and modules. Right now there's already a nice summary of parameters linked as [cheat sheet](http://www.openscad.org/cheatsheet/). Scope of this projects would be to use this information in extended form and make it available in a more direct way in the editor.

## Details

* Convert the cheat sheet information into machine readable format
* Find a way to generate the existing HTML format based on the core data
* Add context help to editor giving help for built-in functions and modules, e.g. by adding formatted help output to the console window, including the links to further documentation like the language manual on Wikibooks

## Expected Outcome

Cheat sheet is integrated into the application and additional context help for built-in functions and modules is available.

## Project Properties

### Skills

* Programming language is C++
* GUI programming with the Qt framework

### Difficulty

Easy

### Additional Information

* Potential mentor(s): [Marius Kintel](https://github.com/kintel) ([IRC](https://webchat.freenode.net/?channels=openscad): kintel), [Torsten Paul](https://github.com/t-paul/) ([IRC](https://webchat.freenode.net/?channels=openscad): teepee)
* Organization website: https://www,openscad.org/

shubhav99, 2021-03-17:
Hi there 
I want to work on this isssue
Please provid me some guidence to start working on this issue

t-paul, 2021-03-17:
Best way to start would be getting the source code of OpenSCAD and build on the machine you want to do development on. Which OS are you using? Setup is probably easiest on Linux, but in general any of the 3: Linux, macOS and Windows should work.

shubhav99, 2021-03-18:
windows ; sir

t-paul, 2021-03-18:
In that case, the best supported option is using the [MSYS2](https://www.msys2.org/) environment.

shubhav99, 2021-03-18:
i need to install openscad and msys2 on my machine ?
and what next 
i need help here mentor 
this is my first issue ever

t-paul, 2021-03-18:
first issue is fine, you should have done some sort of software development before though. MSYS2 is bringing all the development tools, essentially creating a Unix like environment on Windows. You can install all the needed things like compilers, libraries, applications (even including a pre-built OpenSCAD).
A maybe slightly outdated documentation is at: https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/Building_on_Microsoft_Windows

shubhav99, 2021-03-18:
thanks for the support 
i ll work hard on this

shubhav99, 2021-03-18:
i ve installed openscad and msys2 on my machine.
how to proceed next ?

t-paul, 2021-03-18:
Just follow the build instructions on that document linked above. So going through installing the build dependencies, `git clone` to get the source, running `qmake` to prepare the build and finally `make` to compile.

shubhav99, 2021-03-19:
i followed the instructions very carefully 
my qmake is installed but when i try to run ----
qmake openscad.pro CONFIG -= debug
on MSYS
it gives error cannot find file openscad.pro

t-paul, 2021-03-19:
This should be in the top level directory of the repository created via `git clone`. The folder content should contain the files from the main [OpenSCAD repo](https://github.com/openscad/openscad/).

shubhav99, 2021-03-19:
yes sir you are right 
but when i run the command on MSYS 
or on windows terminal , this gives error of openscad.pro file not found 
although i searched for the file in the folder , and file is located in the correct destination 
but MSYS gives error 
i reinstalled it but no .

shubhav99, 2021-03-21:
sir please help me further or lemme drop this contribution so that i can look for some other issue to solve it and gain some experience .....

t-paul, 2021-03-21:
You have to give some more specific questions, you are not saying what command you are running nor what the error message is. If you see the `openscad.pro` file in the source directory as expected then running `qmake` inside that directory (from the MINGW64 shell which was installed with the MSYS2 installer) should work.