opencax/GSoC #35: Cabinet tool for FreeCAD

yorikvanhavre, 2021-03-03:
## Outline

A cabinet tool

## Details

An Arch/BIM tool to create automatically a wide range of cabinets for kitchens, bathrooms, storage... This project should involve research over possible cabinet types, how they are built in the real world, and what properties they should have.

## Expected Outcome

A parametric cabinet object in the Arch Workbench, which is fully driven by properties, and a toolbar button (FreeCAD command) to create the object. Optionally, a friendly task panel UI to create and modify the object.

## Project Properties

### Skills

The student should have a reasonable knowledge of Python, and a basic knowledge of how FreeCAD and its parametric objects work

### Difficulty

Medium

### Additional Information

* Potential mentor(s): [Yorik](https://www.forum.freecadweb.org/memberlist.php?mode=viewprofile&u=68)
* Organization website: https://freecadweb.org
* Communication channels: https://https://forum.freecadweb.org

yorikvanhavre, 2021-03-03:
Other FreeCAD ideas for 2021 at https://www.forum.freecadweb.org/viewtopic.php?f=8&t=54913

brlcad, 2023-02-23:
FreeCAD is now participating as their own GSoC org, congrats!

closed 2023-02-23 by brlcad

Ashura1010, 2024-03-18:
Hey I'm interested for this idea. How can I contact you guys

t-paul, 2024-03-18:
Find FreeCAD in the [2024 org list](https://summerofcode.withgoogle.com/programs/2024/organizations). This should have all the needed information.

yorikvanhavre, 2024-03-18:
https://wiki.freecad.org/Google_Summer_of_Code_2024