opencax/GSoC #36: Improve the FreeCAD API documentation

yorikvanhavre, 2021-03-03:
## Outline

Work on the FreeCAD doxygen-generated documentation: Propose a better plan, document the modules better, make it clearer to read, etc.

## Details

The API documentaiton of FreeCAD is generated with doxygen from the docstrings contained in the source code. It is hosted on https://github.com/FreeCAD/SourceDoc . It is currently not easily readable, the modules structure doesn't list classes and functions, python functionality is not well distinguishable from C++ functionality, and many other problems.

## Expected Outcome

Identify problems and possible solutions, and propose changes to the docstrings and in-source doxygen instructions to build better docs, and possibly do some css work to produce a cleaner HTML result

## Project Properties

### Skills

The student should have or build a good knowledge of doxygen, optionally be ready to do some css work too

### Difficulty

Medium

### Additional Information

* Potential mentor(s): [Yorik](https://www.forum.freecadweb.org/memberlist.php?mode=viewprofile&u=68)
* Organization website: https://freecadweb.org
* Communication channels: https://https://forum.freecadweb.org

yorikvanhavre, 2021-03-03:
Other FreeCAD ideas for 2021 at https://www.forum.freecadweb.org/viewtopic.php?f=8&t=54913

yorikvanhavre, 2022-02-02:
See the new documentation effort at https://github.com/FreeCAD/FreeCAD-documentation which includes auto-generation of API pages

brlcad, 2023-02-23:
FreeCAD is now participating as their own GSoC org, congrats!

closed 2023-02-23 by brlcad

Ovalelephant35, 2024-02-21:
Hey , My name is Anurag Singh I am studying Computer Science in BITS pilani , india , I am interested in this Project I also have some experience in documentation formatting in general if @yorikvanhavre sir you can guide me further it will be helpful
thankyou and regards
https://github.com/Ovalelephant35

yorikvanhavre, 2024-02-22:
Hi @Ovalelephant35 FreeCAD is now handling its own GSoC project. See corresponding issue at https://github.com/FreeCAD/FreeCAD/issues/12547 Please enquire there!