opencax/GSoC #37: Better materials structure in FreeCAD

yorikvanhavre, 2021-03-03:
## Outline

FreeCAD materials are mostly used at the moment by FEM and Arch. There begins to have too many material cards to be comfortable. They need to be organized better.

## Details

Currently, Arch, FEM and Render workbenches are already using the FreeCAD materials system, which is built on "material cards". However, the large number of current material cards turns it cumberstone to use in Arch, FEM, and in the own material editor. We need a better way to classify and select them.

## Expected Outcome

Invent a handier way to select materials, for example grouping them by category. Implement that system in Arch, FEM, Render and the material editor.

## Project Properties

### Skills

The student should have a good knowledge of Python, be ready to learn some Qt/PySide, and have learned how materials work in FreeCAD's Arch, FEM and Render workbenches, as well as the material editor.

### Difficulty

Medium

### Additional Information

* Potential mentor(s): [Yorik](https://www.forum.freecadweb.org/memberlist.php?mode=viewprofile&u=68)
* Organization website: https://freecadweb.org
* Communication channels: https://https://forum.freecadweb.org

yorikvanhavre, 2021-03-03:
Other FreeCAD ideas for 2021 at https://www.forum.freecadweb.org/viewtopic.php?f=8&t=54913

KeithSloan, 2022-02-14:
I would like to see any improvement in materials structure also take into consideration GDML and Geant4 materials.
At present we have to implement own materials.

brlcad, 2023-02-23:
It would be awesome to come up with an interchange format for sharing materials between FreeCAD and BRL-CAD.  This past year, BRL-CAD acquired first class material objects that can encompass any collection of physical, optical, chemical, mechanical, and other properties.

brlcad, 2023-02-23:
FreeCAD is now participating as their own GSoC org, congrats!  (Still open to collaboration on this topic, though tbd)

closed 2023-02-23 by brlcad

davesrocketshop, 2023-04-07:
I would also like to have the ability for external workbenches to add their own directory(ies) of materials for integration into the material handling system.

The Rocket Workbench for example adds materials for rocket components based on manufacturer specs which may differ from the stock materials. These are updated as the manufacturers update their information. For most it's just the denisty property, but  a simple material such as Balsa may have many entries. It doesn't make sense to include all of these in the system materials folder

davesrocketshop, 2023-04-07:
Some usability enhancements:
Currently you can edit values. If you select OK, those values aren't saved, as a save is a separate operation. It just exits discarding the changes. The dialog should confirm whether the user wants to exit or save the changes

When saving, the folder defaults to the system materials folder. It should default to the user defined folder when specified, or the default user folder in FreeCAD.getResourceDir() + "Mod/Material"

KeithSloan, 2023-04-07:
The GDML workbench resorts to using its own materials, it would be good if the FreeCAD facilities allowed for such materials to be supported see [https://forum.freecad.org/viewtopic.php?t=46699](url)

Also in view of the above and additionally it would also be good if FreeCAD had a facility to access Material Properties defined by NIST and in their Database. NIST being the American National Institute.

It will take some good design to allow for the range of Materials and properties used in Arch, FEM, GDML, Rocket Workbench and others and not overwhelm the user with the number available in any selection.

Plus allowing for new properties. i.e. Thermal properties in Arch materials. FEM properties of 3D printed materials.

yorikvanhavre, 2023-04-07:
The materials system in FreeCAD is pretty open, simple and extensible.. I think it would require very few changes to support additional directories and other kinds of sources.

davesrocketshop, 2023-04-07:
Agreed and have started doing so. See https://forum.freecad.org/viewtopic.php?p=673969#p673969

I've added it here though so there aren't multiple people working at cross purposes

KeithSloan, 2023-04-07:
> The materials system in FreeCAD is pretty open, simple and extensible.. I think it would require very few changes to support additional directories and other kinds of sources.

GDML Materials are a bit of a challenge, more so than other Materials that FreeCAD might want to access.

At least one of the issues I found, was in controlling which Materials are selectable. If I was dealing with a GDML Object I really only wanted to offer GDML Materials.

GDML Materials are a hierarchy of
Elements, Isotopes
[image]

Need facilities to change/edit Materials and only produce valid combinations.

Example of an Element 
[image]

Even just offering the standard Geant4 definitions to keep the list of available materials to a reasonable length I broke them into the standard five groups
[image]

Example of a Material
[image]

Details of Geant4 and other standard Materials being availble in an NIST database.