opencax/GSoC #38: FreeCAD command-line preferences manipulation

yorikvanhavre, 2021-03-03:
## Outline

Command-line tools to manipulate FreeCAD preferences from outside FreeCAD, so one could install & configure fully via script.

## Details

Currently it is possible to manipulate FreeCAD preferences options via the Edit->Preferences menu, or via the parameters editor found under the Tools menu. It would be handy to add a way to edit these parameters from scripts, so a FreeCAD installation could be fully automated, including setting preference options. One way to do it could be by adding other command-line options to the FreeCAD executable (which must be done in C++), another maybe more flexible way could be an external command-line tool (which can be coded in any language, but preferably Python), that would create/manipulate the .cfg files where FreeCAD preferences are stored.

## Expected Outcome

A command-line tool that allows to create, modify or delete FreeCAD preference parameters

## Project Properties

### Skills

The student should have a good knowledge of Python or C++, and learn how the preferences system of FreeCAD works

### Difficulty

Medium

### Additional Information

* Potential mentor(s): [Yorik](https://www.forum.freecadweb.org/memberlist.php?mode=viewprofile&u=68)
* Organization website: https://freecadweb.org
* Communication channels: https://https://forum.freecadweb.org

yorikvanhavre, 2021-03-03:
Other FreeCAD ideas for 2021 at https://www.forum.freecadweb.org/viewtopic.php?f=8&t=54913

yorikvanhavre, 2022-02-02:
Since this was written, a new preferences pack system has been added to FreeCAD, that might give new options.

luzpaz, 2022-03-16:
Perhaps the orientation can also for sysadmins deploying FreeCAD in an organization?

brlcad, 2023-02-23:
FreeCAD is now participating as their own GSoC org, congrats!

closed 2023-02-23 by brlcad