opencax/GSoC #39: A new FreeCAD Library tool

yorikvanhavre, 2021-03-03:
## Outline

Design a FreeCAD library tool that fetches the content online, without the need to download the whole library

## Details

FreeCAD possesses a vast library of parts submitted by users, hosted on https://github.com/FreeCAD/FreeCAD-library . This library is huge (> 1.5Gb) and currently the only way to use it is by installing (downloading) it entirely from the Addons Manager. We need a new tool that is able to access both an offline or online version of the library (downloaded on your computer or hosted on a git platform such as github or gitlab) and allow users to browse the contents and download and place objects in their current model. The tool shouldn't depend on a sole git platform as this library could be moved to some other platform in the future.

Optionally, it could also allow users to upload models.

## Expected Outcome

A new FreeCAD UI tool

## Project Properties

### Skills

The student should have a good knowledge of Python or C++, and learn how to build interface tools with Qt/PySide/QtCreator

### Difficulty

Medium

### Additional Information

* Potential mentor(s): [Yorik](https://www.forum.freecadweb.org/memberlist.php?mode=viewprofile&u=68)
* Organization website: https://freecadweb.org
* Communication channels: https://https://forum.freecadweb.org

yorikvanhavre, 2021-03-03:
Other FreeCAD ideas for 2021 at https://www.forum.freecadweb.org/viewtopic.php?f=8&t=54913

yorikvanhavre, 2022-02-02:
@chennes has formulated some ideas on this question, be sure to check with him if interested in working on this

yorikvanhavre, 2023-02-09:
This topic might become part of a bigger project sponsored by the FPA

brlcad, 2023-02-23:
FreeCAD is now participating as their own GSoC org, congrats!

closed 2023-02-23 by brlcad

senyasuper, 2023-03-29:
if it will be big enough to split, Im also ready to participate

senyasuper, 2023-03-30:
> Basically what we are after is a replacement for the Part Library. It has
> become basically unusable in its current form.
>
> The idea is to design a new add-on for FreeCAD. It should have:
>
>    1. Well-defined and documented component metadata, downloadable
>    independent of the component CAD data.
>    2. Non-monolithic data downloads: individual components can be
>    downloaded on demand.
>    3. No additional non-pip-installable software necessary for use (e.g.
>    no requirement for git to be installed).
>    4. A GUI that allows browsing and searching the library before
>    downloading the CAD data, based on available metadata.
>    5. Direct insertion of components into a FreeCAD document via both a
>    GUI and via Python scripting.
>    6. User-configurable data sources: the FreeCAD Project will provide
>    one such source, but that source may be replaced or augmented by other
>    sources.
>    7. The existing parts in the existing Parts Library addon must be
>    imported into the new system and must conform to the new metadata and data
>    storage standard.
>
> That's basically what we would like to obtain.
>
> There is another part to the problem: Imagine what to do with the current
> assets in the Parts Library github repo. It's just too huge to be managed
> by Git. What should we do with that content? Host it ourselves somewhere
> else? Use an existing management system? Create something on our own?
> That's whole part of the story that would make a good GSoC to explore.
>
> This whole content can be tailored into a GSoC as you see fit: A big GSoC
> project that proposes to do it all, or concentrate on a part of it, you
> guys decide. The important is to find and tailor a project scope that fits
> into what you think you can do, and, above all, finding a subject that
> thrills and motivates you and will be fun to work on.

Thanks for providing this information, some of these requirements sound a
bit complicated for me, but about a few of them I have some ideas of
implementation. So first question: is anyone already working on it or are
you waiting for official submission on GSoC? And the library that is
supposed to be implemented in FREECAD is this:
https://github.com/FreeCAD/FreeCAD-library? Also if I'm thinking of taking
175h hours of work, how many of those requirements are supposed to be done
by myself?

yorikvanhavre, 2023-03-30:
@chennes is interested in working on this subject. I did the original setup of the library, and at the moment nobody is working on it. I also don't know of other GSoC proposals for it.

Indeed the whole "server" part is complex. But you could start working on the general design of a new library tool in FreeCAD, that would be able to take its sources from different platforms, and use our current GitHub library repository for now.

The actual migration of that content could be left out of your proposal, and be the matter for another later project.

It is not really a problem if you feel you don't have the knowledge to implement something. Many of us can help you with that. It's more important here to be ready and interested in exploring, communicating, and working with us to arrive to a good plan. If you are interested in working on this, I would basically concentrate on designing how such an application should work, deigning/mockuping UI components and behaviours, etc, and implementing up to where you will be able to go. In GSoC projects, delivering a finished result is not a mandatory point. The important is to dedicate work on the subject, and help the project to advance in that direction.

senyasuper, 2023-03-30:
Yes, I'm interested in it, is FreeCAD forum an appropriate place to introduce myself and submit first version of proposal for feedback or there are other options to contact you?

yorikvanhavre, 2023-03-30:
> (By "interested in working on this subject" @yorikvanhavre means that I am available to mentor you, not that I want to do it myself

Yes poor choice of words, sorry :sweat_smile: