opencax/GSoC #41: Add the ability to the BCF library to connect to BCF APIs

Moult, 2021-03-09:
## Outline

The "BCF" system, or BIM Collaboration Format system is an international standard to track issues in building design and coordination. It's similar to issue tracking on software projects, just like this one, so it stores things like issue titles, assigned responsibilities, tags, and issue comments. It also contains discussions about issues in 3D building models, like section planes coordinates, highlighting 3D geometry, and camera views. There are two ways BCF issues can be exchanged - either manually through files ending in the .bcf extension, or via an API.

There is already a library which handles reading and writing these files through the .bcf extension, but not via the API. This is where we need your help! Help build one which connects to this REST API.

## Expected Outcome

The existing bcf library will get the ability to connect to BCF APIs and read and write BCF related data.

## Future Possibilities 

Integration into Blender or FreeCAD

### Skills

 - Python knowledge
 - Knowledge about how to connect to APIs, including how a REST API works
 - (Optional) basic Blender or FreeCAD experience

### Difficulty 

Easy

### Additional Information

* Potential mentor(s): Dion Moult
* Organization website: https://ifcopenshell.org
* Communication channels: https://community.osarch.org , ##architect on Freenode , and https://github.com/IfcOpenShell/IfcOpenShell/issues

TestPrab, 2021-03-15:
Hi, I am interested in working on this issue , could you point me to some commit or some docs with which I can start.

Moult, 2021-03-16:
@TestPrab g'day! I've added some doc links - let me know if there are any issues. I reckon the first thing to do to start is to play with the bcf library, and download some test cases from the BCF-XML repo to get a feel of what a BCF is. You might also want to install the blenderbim.org add-on to Blender, as it will let you manipulate BCF in a graphical environment, which is required to test the graphical aspects of the BCF spec.

Let me know if anything is confusing, and we should have an introductory chat too! I'm online in the Freenode chat channel!

Moult, 2023-02-25:
Closing as now been implemented.

closed 2023-02-25 by Moult