opencax/GSoC #42: System to generate construction sequencing animations

Moult, 2021-03-09:
## Outline

Often, construction firms need to visualise animations of construction sequencing. A project timeline will be created, and related to individual model elements. For example, when a concrete slab is poured, it is linked to a 3D object called a slab. We need the ability to automatically generate animations from Blender where objects appear / disappear in various different ways when they start / end their task in the project timeline.

## Details

## Expected Outcome

A series of small scripts that take objects and can automatically animate the visibility, locations, or staggered appearances of building elements, as well as sub elements, and basic scripts that correlate real world time to animation frames, and frames per second, and generate an animated timeline bar in various styles.

## Future Possibilities 

This animation system can be then used from BIM models either in Blender, FreeCAD, or via other software altogether, so it has quite a large impact on the ecosystem.

### Skills

 - Basic knowledge of the principles of animation (keyframing)
 - Basic Blender animation (you can do some tutorials and get up to speed pretty quick)
 - Python
 - Artistic sense! We should offer beautiful and elegant animations!

### Difficulty 

Easy

### Additional Information

* Potential mentor(s): Dion Moult
* Organization website: https://ifcopenshell.org
* Communication channels: https://community.osarch.org , ##architect on Freenode , and https://github.com/IfcOpenShell/IfcOpenShell/issues

Szymon-Budziak, 2021-04-07:
Hello, I am interested in working on this project for GSoC. Is is still available?

Moult, 2021-04-07:
@Szymon-Budziak yes, this is still available, but you've got to be quick as the deadline is 14th of April. Catch me online on IRC via the OSArch.org live chat (##architect on Freenode) and let's discuss! Also, get a head-start and start writing your proposal!

samadpls, 2023-02-25:
I am interested in working on this project as it aligns with my experience. May I inquire about the best way to reach you for further discussion?

Moult, 2023-02-25:
Absolutely! Feel free to reach out on osarch.org/chat and/or dion@thinkmoult.com and let's have a chat further!

jaymeandretta, 2023-03-09:
Nice feature! I'd like to participate! I've never applied for GSoC. Is there anything I should do before applications start? @Moult

Moult, 2023-03-09:
@jaymeandretta indeed! Reach out on osarch.org/chat - introduce yourself, start looking around the codebase and familiarise yourself with the code!

niuniuzhao, 2024-03-14:
Hello everyone, Hi @Moult ! I am an architecture student and a GSoC2024 applicant. I'm very interested in this project because it seems like a great fit! Could you please advise whether the scope of the project would be within the reach for someone with sophomore computer science knowledge - wide introduction + data structures (and a lot of motivation, of course)? I love a challenge but also don't want to overshoot too much. :) Thanks!

Moult, 2024-03-15:
G'day @niuniuzhao ! Yes indeed, this is a pretty straightforward GSoC project! It would be great to start off with a 1-page summary of any thoughts you had about how to tackle this and what you'd like to build, and when you're ready let's meet up for a voice call!

niuniuzhao, 2024-03-24:
Hi @Moult , I hope you are feeling better! I have done up a 1 page summary, and it would be great if we could do a quick call, maybe between 10am-1pm Sydney time? Thanks!

Moult, 2024-03-24:
@niuniuzhao absolutely! I'm free today if you'd like, just join https://osarch.org/chat and let's meet up!

creativechirag26, 2024-08-24:
Hey @Moult , 

I am Chirag and I went through the project idea, I know its a GSOC Project, but if you don't mind, I would like to gain some knowledge by contributing into this project.

Moult, 2024-08-24:
Dion Moult

Sent from Proton Mail mobile

-------- Original Message --------
On 24 Aug 2024, 4:21 pm, Chirag Ingle wrote:

DraKen0009, 2024-09-01:
Hello @Moult  , I would like to contribute to issue, since it was the only easy one with python requirement. I haven't really worked with animation specifically but I'm decently good with python so I can learn few new things with it working on it. 
Also can you please guide me to any contributor-guide available for the org and where can I find issues related to python. 

Thanks for time.

Moult, 2024-09-01:
G'day! https://docs.bonsaibim.org/ and https://github.com/IfcOpenShell/IfcOpenShell/issues should help!

basvi-chunara, 2024-12-23:
Hello @Moult , I would like to know if this project is still available to work on? 
I'm really looking out for projects like these to increase my open-source knowledge. Any advice for a beginner ? Should I directly start with project or work on some good-first-issues ? 
Thank you for your time!

Moult, 2025-03-06:
@basvi-chunara g'day, yes, this is still available.

leonstrr, 2025-03-07:
Hey @Moult! 
I am interested in this issue, too. Is it generally foreseen that more than 1 contributors work on (this) issue(s)?
Looking forward hearing from you!

Moult, 2025-03-07:
@leonstrr in general there is only one contributor per project. The only time you have 2 contributors is if the project can be divided into two with clear scope for each :)

Would you be interested in applying?

basvi-chunara, 2025-03-07:
Hello @Moult, thank you for your response!
I'm interested to work on this project and planning to apply to gsoc 2025 to gain hands-on experience:)
Also, I am open to collaborate (if this project can be divided with two scopes!) 

Would like to know if you've any tips for a strong gsoc application? 
Looking forward to your insights:)

Moult, 2025-03-07:
Thats great! The strongest application is where the applicant has already actively reached out on our chatroom https://osarch.org/chat and has already installed and tried our software so they know the context of what they're building, and has already written a little bit of code related to the project in some way so that they aren't going in blind, they know what's involved.

More tips here https://opencax.github.io/

Dion Moult

basvi-chunara, 2025-03-07:
Got it! Thanks a lot @Moult

AmandeepMandal1077, 2025-03-14:
Hi, i would like to know more about this issue.
From what I understand, the task is to develop a set of basic animation scripts that can be applied to models. Essentially, it's pre-built animations—such as bounce, fade in/out, appear, and disappear—that can be easily added to any basic shape model.
am i right.
I would appreciate any additional details or guidance.

Moult, 2025-03-14:
G'day! The best thing to do is to reach out in the osarch.org/chat and
say hello. Let us know your experience in Blender, or animations, and
Python :)