opencax/GSoC #43: Create visual programming nodes for generating BIM data with IfcSverchok

Moult, 2021-03-09:
## Outline

Blender allows visual node programming, similar to Grasshopper in Rhino. Grasshopper has a really neat extension called GeometryGym which allows users to use visual node programming to generate building geometry and data using the "IFC" international standard. It's super awesome, and something like that doesn't exist yet in the open source world with only free software. So, let's create it!

Some of this is already started, so the basics are already coded, but now it needs to be tested, debugged, and a whole bunch more nodes written and tested out in experiments in generating buildings with visual nodes.

## Expected Outcome

A whole bunch more nodes, and little examples of generating buildings via nodes, and getting an IFC output.

## Future Possibilities 

If we code it right, in the future, these nodes can be made agnostic of Blender and also work in FreeCAD (e.g. through PyFlow). So this allows multiple authoring apps to benefit from visual programming.

## Project Properties

### Skills

 - Python
 - Knowledge of visual programming (knowledge with Grasshopper / Sverchok super useful!), if not, get ready to watch tutorials on them!
 - Basic Blender knowledge

### Difficulty 

Medium

### Additional Information

* Potential mentor(s): Dion Moult
* Organization website: https://ifcopenshell.org
* Communication channels: https://community.osarch.org , ##architect on Freenode , and https://github.com/IfcOpenShell/IfcOpenShell/issues

nortikin, 2022-06-29:
watching

t-paul, 2022-06-29:
@nortikin you can use the `subscribe` button in the side bar to request updates on the issue without sending out mail notifications to everyone already subscribed already.

Moult, 2023-02-28:
For anybody interested in tackling this project for GSoC 2023, you would be picking up where @mdjska left off and continuing to develop the nodes further!

nortikin, 2023-03-01:
> For anybody interested in tackling this project for GSoC 2023, you would be picking up where @mdjska left off and continuing to develop the nodes further!

does it mean it is no developer/support anymore?
Just trying out ifc on windows, not handle as user, learning. Maybe make me contributor?

Moult, 2023-03-01:
@nortikin done :)

Source code is here https://github.com/IfcOpenShell/IfcOpenShell/tree/v0.7.0/src/ifcsverchok

andrsbtrg, 2023-03-10:
hey @Moult ! I am an interested in tackling this one for GSoC. I have developed before nodes for grasshopper and have knowledge of BIM, and would like to contribute to the project. Is it still open?

Moult, 2023-03-10:
@andrsbtrg yes this is still open! Feel free to introduce yourself over at osarch.org/chat and start playing around with the current codebase linked above!

mdjska, 2023-03-11:
@andrsbtrg would be cool if you continued the project, if you have any specific questions about the codebase feel free to reach out!

keshavanarayan, 2024-03-08:
Hi @Moult Is this porject being worked on currently? If not, I'm interested.

Thank you

Moult, 2024-03-08:
@keshavanarayan nothing is being actively worked on for GSoC 2024 as we're still in the idea discussion phase. See the timeline here: https://developers.google.com/open-source/gsoc/timeline

The best way to get started is to get IfcSverchok running and see what works and what doesn't, and start planning ideas on where to take it next.

chrisjesuscj, 2025-03-18:
Hi @Moult . I'm interested in working in this project for GSoC 2025. I'm an architect and long time user of sverchok and I've been learning python for over a year now, having created a python lib for my master in BIM at PUC-Rio. Please let me know if this project slot is still open.

Moult, 2025-03-18:
G'day! Yes absolutely all slots are open right now. Check the GSoC
timeline for more details. The best thing to do is to join
https://osarch.org/chat and say hi :)

-- 
Dion Moult