opencax/GSoC #44: Create a webapp to create and check BIM project exchange requirements

Moult, 2021-03-09:
## Outline

When projects exchange data, we often need to set contractual requirements about what data we expect to see in their CAD or Building information data. There are international standards for these, and existing tools we've written to check these requirements. However, they look very scary and don't lend well to users understanding how to create project requirements. This ends up with a situation where many people manually check requirements, or rely on expensive proprietary software that only a few people can check requirements with. No fun.

A little mockup of what a webapp would be like is: https://blenderbim.org/bimtester/

In short, there are templates of exchange requirements, which we need to store and organise in a standard way which hasn't yet been decided. Descriptions of each requirements need to be stored. It needs to create a simple text file storing the requirements. It needs to link to pages describing which apps can meet these requirements. It needs a button to import requirements, or import custom requirement definitions. Finally, we need to check these requirements online!

Users upload their building model in the IFC format. This can come from either Blender or FreeCAD, or something else entirely. There is already an app called BIMTester which does the checks, but no web UI.

## Expected Outcome

A working example of the proof of concept webapp.

## Future Possibilities 

So much!

## Project Properties

### Skills

 - Webby stuff (HTML, JS, and either PHP or Python as a backend, although PHP is preferred as it reduces hosting barriers for contributors)

### Difficulty 

Medium

### Additional Information

* Potential mentor(s): Dion Moult
* Organization website: https://ifcopenshell.org
* Communication channels: https://community.osarch.org , ##architect on Freenode , and https://github.com/IfcOpenShell/IfcOpenShell/issues

sky3760000, 2021-03-17:
Hi I m intrested and little confused . Is this ideas for cad organizTio

brlcad, 2021-03-18:
@sky3760000 Do you have a more specific or more detailed question?  Clear communication is important.  What you wrote was not clear at all.  This is a project idea for our umbrella organization, yes.

sky3760000, 2021-03-23:
Hi, 
My questions are as follow-:
1 .Whether the webapp should strictly look like same as the example given or changes are welcomed.
2. The web app should be more css oriented(like design) or javascript(fucntions)
3 Do you have any specific template for proposals
 Thanks for your reply 
My email-akashpandey382@gmail.com

Moult, 2021-03-23:
Hey @sky3760000 - a student has already started work on this - perhaps we can discuss to find another task that might be suitable for you? What aspects of the AEC industry are you interested in?

sky3760000, 2021-03-24:
No problem,
What is AEC industry?

Moult, 2021-03-25:
@sky3760000 I think it makes more sense to discuss for an alternative thing you can work on. What are your interests? What type of project / tech might you like to hack on? Would you like to organise a call to discuss your ambitions? Do you want to build web apps? Do you want to work with APIs? Do you want to work with 3D graphics? Write mesh algorithms? Do format conversions and parsing (i.e. import/export)?

AEC = Architecture, Engineering, and Construction. The IfcOpenShell project is primarily used in the AEC industry.

sky3760000, 2021-03-25:
I m intrested in web apps and also want to learn api so can u suggest any vacant project which I can work on.

Moult, 2021-03-26:
For web app related things, there may be a task in porting IfcOpenShell to WASM. Is that something you'd like to tackle?

Outside web apps, which form the bulk of the work, perhaps you may be interested in https://github.com/opencax/GSoC/issues/31 ?

Hemanth694, 2021-04-11:
Hey!
So I am intrested in creating webapps too so can you just help me out?
Like should I make a proposal for this project or is there any other step to go further?

Moult, 2021-04-11:
@sky3760000 sorry for the late reply, I did not see your edit. What tips are you after?

@Hemanth694 g'day! We're a bit close to the proposal deadline, but this web application idea already has a student proposing for it. Is there another aspect of IfcOpenShell that interests you?

Hemanth694, 2021-04-11:
Another aspect of IfcOpenShell?.... I did'nt quite get you.. you mean another project?
Is there any chance?

Moult, 2021-04-11:
@Hemanth694 certainly, BIMTester is merely a module of IfcOpenShell

Hemanth694, 2021-04-11:
Ohkay
Well then I'm also interested in #41

Moult, 2021-04-11:
@Hemanth694 hmm, that one already has a proposal. How about either #40 or #31 - both of which do not currently have proposals?

Alternatively, you can come up with your own proposal!

Hemanth694, 2021-04-11:
Alright Lemme check those and I'll try anyone of them
Thnx for the response Appreciate it.

Abdo865, 2023-02-28:
Hello I would like to contribute on this ?
can anyone help me were to start?

brlcad, 2023-03-01:
> Hello I would like to contribute on this ?
> can anyone help me were to start?

@Abdo865 This project idea has not been tagged for GSoC 2023, so I suggest looking at the project ideas list again at https://opencax.github.io

Moult, 2023-03-01:
Apologies, I have already updated this to be relevant to GSoC 2023. Here are the first three steps I would advise for anyone interested in this:

 1. Get this webapp running on your machine: https://github.com/IfcOpenShell/IfcOpenShell/tree/v0.7.0/src/ifctester/webapp - it's a simple static site with no need for any backend.
 2. Read this documentation - the table of contents has 10 sections https://github.com/buildingSMART/IDS/tree/master/Documentation
 3. Reach out at osarch.org/chat and say hello! Ask your questions, meet up for a short voice call, introduce why you're interested and whether the code makes sense, and discuss a plan collaboratively for the next steps!

Abdo865, 2023-03-01:
Hello, is this project is still available in 2023 project list to contribute?
If it's still available, can you suggest me a code challenge to try in it to be accepted to contribute with you?

Moult, 2023-03-01:
@Abdo865 sure, see the three steps I posted above. If the code makes sense to you, we can agree on a small change to be made.

hiri09, 2023-03-03:
@Moult hi sir ,
can you explain me the 2nd step like which file i have to use as documentation

Moult, 2023-03-03:
@hiri09 the link I posted in step 2 is the documentation: https://github.com/buildingSMART/IDS/tree/master/Documentation - you can follow the beginners tutorial and read the 10 sections in the  "Begin learning about IDS" section.

hiri09, 2023-03-04:
@Dion Moult I have read those 10 section and also I got so many things as
well so can you guide me like what do I need to do

hiri09, 2023-03-04:
@Moult I read the documentation you provide and I think this like checking the IFC files through IDS and also got like facet and all but can you tell like how and what I have to make in webpage?

Moult, 2023-03-04:
@hiri09 have you tried running the web app in step 1? Did you try it out with the tutorial in step 2?

Moult, 2024-01-21:
Note to anybody reading this that this issue is available for GSoC 2024!

Apurv428, 2024-02-11:
@Moult, I'm keen on participating in GSoc 2024 for this project. Could you provide some assistance to help me begin?

dion, 2024-02-11:
don't @ me anymore

Moult, 2024-02-11:
@Apurv428 it's a bit early in the timeline https://developers.google.com/open-source/gsoc/timeline - can you wait until Feb 21 then message again depending on whether or not this organisation has been approved for GSoC?

BalrajDhakad, 2024-02-22:
Hey, 
this is Balraj,
I am keen to work for this project and I am also looking forward for any guidance I can get in order work for this project and also get into GSOC 2024. @Moult

Moult, 2024-02-27:
Hey @Apurv428 @BalrajDhakad  ! The organisation has been approved so now is an excellent time to make a start. The checklist is here: https://opencax.github.io/gsoc_checklist.html and you can reach out at https://osarch.org/chat

dion, 2024-02-28:
stop sending me these messages learn how to @ the correct person

keshavanarayan, 2024-03-08:
@Moult is this being actively worked now for GSoC 2024? If not I'm interested.

Moult, 2024-03-08:
@keshavanarayan nothing is being actively worked on for GSoC 2024 as we're still in the idea discussion phase. See the timeline here: https://developers.google.com/open-source/gsoc/timeline

If you're interested, join in the chat and start writing a draft application! You'll find three beginner steps to get started above.

Kenyg369, 2024-03-13:
Hi @Moult 
I am a former structural designer, civil engineer, and now a new graduate developer. I am very very interested in this project!!
I have joined the chat, read the guidance, and started working on my draft application. I wonder if it's ok to let you share some ideas after I am done with my draft.

Moult, 2024-03-14:
Hi! Yes please reach out for a chat e.g. a voice call during the Sydney timezone ideally or with Vukas Pajic in the chat when you're ready to discuss ideas!

The webapp is intended to allow people to create and edit IDS files. Auditing is a nice bonus.

Dion Moult

Kenyg369, 2024-03-14:
> Hi! Yes please reach out for a chat e.g. a voice call during the Sydney timezone ideally or with Vukas Pajic in the chat when you're ready to discuss ideas! The webapp is intended to allow people to [...]

Thanks for such a prompt response!! Do you have any time available this weekend? I am flexible with time. Let me know if you are available. Much appreciated!

Moult, 2024-03-14:
I'm currently recovering from sickness but how about you start by writing a 1 page summary of your thoughts and we can organize early next week? Say Tuesday. Can you do between 9 to 5 Sydney time?

Dion Moult

-------- Original Message --------

Kenyg369, 2024-03-15:
@Moult Hope you get well soon!!! Yes, absolutely! I will try to do that summary now. I just have many things on this project I am not clear about.

shwet91, 2025-01-04:
hi...I am proficient in html , css , javascript ,

this is the project I have made : https://shwet91.github.io/snake/

soniya-malviy, 2025-01-05:
@Moult sir  Hi..
My Tech stack:
Javascript , AWS Cloud computing , MERN Stack 
can you please suggest a project where i can contribute 
This is my project:[](https://cure-connect-full-satck.vercel.app/)(Wait for 2-3 min to load data from database)