opencax/GSoC #45: Add IDS checking to ifcOpenShell

berlotti, 2021-03-10:
## Outline
The buildingSMART standard to define IFC based 'Information Delivery Specifications' (IDS) has a first version (and probably a final version around summer). It would be cool to have the ability to validate IDS against an IFC file.

## Details
The IDS standard is being developed and worked on here https://github.com/buildingSMART/IDS/blob/master/Development/

## Expected Outcome
Add IFC file + IDS file and get check result back (preferably in BCF)

## Future Possibilities 
Only limited by your imagination.... 

## Project Properties

### Skills
XML, XSD restrictions and patterns. 

### Difficulty 
Easy to medium (depending on knowledge about IFC)

### Additional Information

* Potential mentor(s): buildingSMART International
* Organization website: https://github.com/buildingSMART/IDS/
* Communication channels: technical@buildingsmart.org

Moult, 2021-03-10:
I believe this is already possible as part of BIMTester, sans the BCF output, but addition of BCF support I believe is too small a task for GSoC. https://github.com/IfcOpenShell/IfcOpenShell/issues/1349

atomczak, 2021-04-08:
As discussed I would like to give it a try and further develop this feature by:
- [ ]  developing and making sure IDS checking works against new XSD
- [ ]  creating some IDS samples and wiki to make it easier to start for new people from AEC like myself :)
- [ ]  add the BCF output of such report
- [ ]  add IDS authoring and integrate with BIMTester web UI 

I'm submitting the GSoC proposal, also available here: [https://docs.google.com/document/d/15oQDhYrApuHsI1Zee5bDvSqRoUz67xHMr79hdFJUoZ8/edit?usp=sharing](url)

[image]

Moult, 2021-04-09:
Great! Let's continue the conversation over in https://github.com/IfcOpenShell/IfcOpenShell/issues/1349

closed 2022-02-19 by aothms