opencax/GSoC #46: Create a new importer

mikwielgus, 2022-01-19:
Outline
-------

Create a new importer for files from any EDA software for which there is
no built-in importers yet.

Details
-------

A non-exhaustive list of possible EDA software:

- Easy (open source): Fritzing, gEDA, Horizon EDA, LibrePCB, pcb-rnd
- Easy, but slightly harder (port existing importer): EasyEDA
  ([link](https://github.com/wokwi/easyeda2kicad)), Zuken CR-5000
  ([link](https://forum.kicad.info/t/zuken-cr-5000-to-kicad-converter/22923))
- Medium (text formats): Allegro ASCII, PADS ASCII, Mentor HyperLynx
- Hard (binary formats): Target 3001!, DipTrace, EAGLE < 5.91, Allegro,
  Mentor

There are two FOSDEM talks around this topic to get a rough idea of the task:

- [Importing into KiCad from CADSTAR](https://archive.fosdem.org/2021/schedule/event/openhard_cadstar/)
- [Reverse-Engineering of (binary) File-Formats](https://archive.fosdem.org/2021/schedule/event/reverse_engineering/)

In case you want to write an importer for a proprietary EDA program, you need to have legal access to this software! Either there is a demo or student version available, or you have some other means of access, like your university institute (please ask them beforehand). We don't want someone to get into legal trouble. If you are unsure about this topic, choose one of the open-source tools.

Expected Outcome
----------------

An importer for files from EDA software for which there was no built-in
importer. Depending on the complexity, this might include board, schematic and/or library import.

Project Properties
------------------

### Skills

- C++
- Basic knowledge of the EDA software the importer will be for

### Difficulty

Depends on the selected EDA package.

### Size

350 hours

### Additional Information

- Potential mentors:
    - [Roberto Fernandez Bautista](mailto:gsoc@kicad.org)
    - [Thomas Pointhuber](mailto:gsoc@kicad.org)
- Organization website: https://kicad.org
- Communication channels:
    - gsoc@kicad.org
    - Real-time communication channels with KiCad developers will be available

zingo, 2022-03-11:
For EasyEDA there is also https://github.com/RigoLigoRLC/LC2KiCad

Trashman247, 2022-04-11:
Hello,

I have a strong interest in logic/circuit design and this project interests me for the GSoC. Just found out about GSoC and realized the deadline is coming soon, so I just wanted to get this comment in to notify as early as possible. I plan to comment soon with a more in depth proposal: I hope to get advice on crafting a project proposal for April 19th then! Please let me know if this works.

Thanks!

mikwielgus, 2022-04-11:
Sure, go ahead. I recommend you to discuss your proposal with us on the Discord channel or via email before submitting.

zingo, 2022-04-14:
Trashman247 If you (or anyone else) do this ping me (here or zingo@zingo.org) and I can package some current progress of my EasyEda json import try for you as as something to look at or get inspired of, or to use use base :).
I currently got to it to the point where the fun begins e.g. I can select import from the menu and select a file and it read in the json. Next step would be to dig out the needed stuff from the json struct and create KiCad stuff from it. My plan was to use the code from https://github.com/RigoLigoRLC/LC2KiCad to figure out the actual conversion (got ok from RigoLigoRLC about it) .

Trashman247, 2022-04-15:
Thanks! I am just entering college and I don't think I have the skill set yet to take on open source. Don't know the future holds, but I'll come back for GSoC 2023! (or if your open to non-GSoC contributions, but it will still take some months for me to learn either way).

closed 2023-03-03 by sethhillbrand