opencax/GSoC #47: Modernize the EAGLE importer

mikwielgus, 2022-01-19:
Outline
-------

The EAGLE importer for schematic and PCB is out of date. It should be
refactored or rewritten to follow the architecture of CADSTAR or Altium
importers where we first load to domain-level objects and then convert
to KiCad objects.

Details
-------

- Rewrite the parser to use domain objects like in the Altium and
  CADSTAR parsers
- Check for correctness and completeness, including current EAGLE
  version features
    - E.g. import EAGLE multiple sheet schematics
      ([issue](https://gitlab.com/kicad/code/kicad/-/issues/1813))
- Add support for Fusion360 import (should be the same data format, just
  inside an archive)
- Write QA tests for all major parts of the importer

Epic: https://gitlab.com/groups/kicad/code/-/epics/40

Expected Outcome
----------------

A modernized EAGLE importer structured like the Altium and CADSTAR
parsers.

Project Properties
------------------

### Skills

- C++
- Basic knowledge of EDA software concepts

### Difficulty

Easy

### Size

175 hours

### Additional Information

- Potential mentors:
    - [Roberto Fernandez Bautista](mailto:gsoc@kicad.org)
    - [Thomas Pointhuber](mailto:gsoc@kicad.org)
- Organization website: https://kicad.org
- Communication channels:
    - gsoc@kicad.org
    - Real-time communication channels with KiCad developers will be available

thegeekymuggle, 2022-01-23:
Hi, I have some experience in C/C++ and python. I am new to the open source scene and would love to contribute to this project. I need help to navigate this. Also, can someone help me to find the IRC channel for the same? Thanks in advance.

t-paul, 2022-01-23:
There is a `#kicad` channel on Libera.Chat, but this is more focused on user discussion. If you want to contribute in scope of GSoC, I'd recomment the email address listed above, otherwise follow the guides on https://dev-docs.kicad.org/en/contribute/

Tejasd1, 2022-08-04:
I Would like to contribute in it....

Tejasd1, 2022-08-04:
I have experience of **C/C++,** **Java** & **Python**

sethhillbrand, 2022-08-04:
@Tejasd1 The GSOC period is closed for this year but if you are interested in working on this or other aspects of KiCad, we are always interested in helping to onboard new contributors.  You can join us over on [Discord](https://discord.gg/WhzyWZ5RaD) and we can help you get started

closed 2023-03-03 by sethhillbrand