opencax/GSoC #49: Port expression evaluators from Lemon to PEGTL

mikwielgus, 2022-01-19:
Outline
-------

Many input fields in KiCad can evaluate arithmetic expressions, taking
their results as the actual inputs. This is accomplished using two
parser-evaluators implemented with the [Lemon Parser
Generator](https://sqlite.org/src/doc/trunk/doc/lemon.html), which we
however intend to replace.

The task is to rewrite these evaluators using
[PEGTL](https://github.com/taocpp/PEGTL), a header-only C++ template
library, and compare them to what is used currently.

Details
-------

- Port the expression evaluator used in unit binders
  ([grammar](https://gitlab.com/kicad/code/kicad/-/blob/master/common/libeval/grammar.lemon))
- Compare the new evaluator to the original one.
- Do the same for the expression evaluator used in DRC rules
  ([grammar](https://gitlab.com/kicad/code/kicad/-/blob/master/common/libeval_compiler/grammar.lemon))
- Write QA tests for both evaluators

Expected Outcome
----------------

Both of the Lemon parsers reimplemented in PEGTL and compared to their
original implementations. QA tests are implemented to check for their
correctness.

Project Properties
------------------

### Skills

- C++
- Parser generators

### Difficulty

Medium

### Size

350 hours

### Additional Information

- Potential mentors:
    - [Mikolaj Wielgus](mailto:gsoc@kicad.org)
- Organization website: https://kicad.org
- Communication channels:
    - gsoc@kicad.org
    - Real-time communication channels with KiCad developers will be available

AhmedTareek, 2023-02-26:
Hi I would like to work on this project as a GSoC participant, Is it still available? 
@mikwielgus

mikwielgus, 2023-02-26:
@AhmedTareek As much I'd love to mentor this, I'm not a KICad developer anymore and I don't know if KiCad is participating in GSoC this year (I suspect not).

lordofbikes, 2023-02-26:
@AhmedTareek you should use the above mentioned Communication channels to get in touch with KiCAD maintainers. Send an email or join their Discord channel.

closed 2023-03-03 by sethhillbrand