opencax/GSoC #52: BRL-CAD Python bindings

drossberg, 2022-02-03:
## Outline
Implement bindings for the BRL-CAD functionality to Python programming language

## Details
There are long time on-going efforts to wrap BRL-CAD functionality with python code, e.g.
* https://github.com/kanzure/python-brlcad
* https://github.com/nmz787/python-brlcad-tcl

These projects are however still in early development stages.

## Expected Outcome
A Python module which can read and write BRL-CAD databases, and provide access to their contents to read, create, and modify the objects.

## Project Properties

### Skills
* C/C++
* Python

### Difficulty
hard

### Size
can be adapted

### Additional Information
* Potential mentor(s):
[Daniel Rossberg](mailto:devs@brlcad.org)
[Sean Morrison](mailto:devs@brlcad.org)
* Organization website: https://brlcad.org
* Communication channels: https://brlcad.zulipchat.com

aniketjohri23, 2022-03-09:
Hello mentors, I am interested in working on this project. Can you give me a starting point to work on this.

brlcad, 2022-03-09:
Hello @aniketjohri23 -- the starting points are listed in the project description.  There are two previous efforts.  The most recent in 2018 has additional information you can read up on at https://brlcad.org/wiki/Google_Summer_of_Code/2018

He published a summary at https://medium.com/@Mr_Jaypee/brl-cads-python-procedural-geometry-990e3c286a63

You'll need to review and understand those prior efforts in order to propose something meaningful.  There are lots of possibilities but one of the most basic is described above -- update/create the ability to read, write, and modify objects using python.  An extension I think would be really powerful is the ability to define a script entity in BRL-CAD that could be described by python (or ged script or lua script, etc).

drossberg, 2023-03-14:
I think, you found our Zulip channel, right?

anirban-1009, 2023-03-14:
yes!

niuniuzhao, 2024-03-14:
Hello there! I am an architecture student and a GSoC2024 applicant. I'm very interested in this project because it seems like a great fit! Could you please advise whether the scope of the project would be within the reach for someone with sophomore computer science knowledge - wide introduction + data structures (and a lot of motivation, of course)? I love a challenge but also don't want to overshoot too much. :) Thanks!

drossberg, 2024-03-14:
Hi @niuniuzhao, thanks for your interest in BRL-CAD. Welcome!

The BRL-CAD Python API could indeed be an easy project, depending on what is easy for you. You should be firm in C, C++ (more passive, i.e. reading), and Python (writing code) and you should have a sense of API design.

If you are still interested, you should familiarize yourself with our software first:
* Install BRL-CAD. You can find binary packages for Windows at https://github.com/BRL-CAD/brlcad/releases. This will install over 200 programs. Most of them are small command line tools. The most interesting for you are our old but still heavily used GUI mged and the more modern GUI archer.
* Go through the mged tutorial https://brlcad.org/w/images/c/cf/Introduction_to_MGED.pdf. The tutorial will show you how to create geometries, assign materials, create ray-trace images, and many more.

For discussions, we have our [Zulip chat](https://brlcad.zulipchat.com).

Parv-gugnani, 2024-03-15:
hello i want to help how can i contribute

drossberg, 2024-03-15:
Hi @Parv-gugnani, welcome! For you applies the same as for niuniuzhao ;)

Shreyansh260, 2025-02-16:
I am Shreyansh Singh Rathore, a B.Tech student specializing in AI and Data Science. I have a strong background in Python and have worked on several projects, including a Python-based Vulnerability Scanner for web applications.

I am highly interested in contributing to the BRL-CAD Python Bindings project for GSoC 2025. I have reviewed the existing efforts, such as (https://github.com/kanzure/python-brlcad) and (https://github.com/nmz787/python-brlcad-tcl), and understand that the goal is to create a Python module for interacting with BRL-CAD databases.

I would love to discuss how I can contribute effectively to this project and refine my proposal based on your guidance. Could you provide insights into the current challenges or specific areas that need improvement? Also, do you recommend any particular resources to get started?

Looking forward to your response.

Best regards,
Shreyansh Singh Rathore

drossberg, 2025-02-16:
Hi @Shreyansh260, thanks for your interest. The organizations for this year's GSoC weren't announced yet. We don't know, if BRL-CAD can take part.

However, you should start with learning about BRL-CAD and how to build it first. You can do this in any order: Install BRL-CAD from a binary release, do the tutorial, and build the programs from sources afterwards. Or, you start with building and installing the programs, and do the tutorial then.

When you know, how BRL-CAD works, the first step of writing a Python binding would be to design the Python API: How shall the calls to BRL-CAD look like in a Python script?

For discussions, we have our [Zulip chat](https://brlcad.zulipchat.com). Please, do any further conversation there.

Spurgeon008, 2025-03-30:
I am John Spurgeon, a B.Tech student specializing in IT with experience in Python, C, and C++. I have worked on various projects involving software development, backend engineering, and database management.
I wanted to ask how I can increase my chances of getting selected for this project. Would contributing before the proposal submission be beneficial, or is there a code challenge I should complete? Additionally, are there any specific areas where contributions would be most impactful?

Looking forward to your guidance.

Best regards,
John SpurgeonI am highly interested in contributing to the BRL-CAD Python Bindings project for GSoC 2025

drossberg, 2025-03-30:
Hi @Spurgeon008, welcome! You should learn about BRL-CAD first: What it is, how to use it, and how to build the programs from the sources. Then, you should sketch your ideas for a Python API and how to implement it in your proposal. This would be enough work for the time till the proposal deadline.

After that date, you should contribute code via a pull request on GitHub to show your skills.

For discussions, we have our [Zulip chat](https://brlcad.zulipchat.com). Please, do any further conversation there.

TanushreeSarkar, 2025-04-09:
Subject: Interest in BRL-CAD Python Bindings Project – GSoC 2025

Dear BRL-CAD Team,

I am Tanushree Sarkar, a third-year B.Tech student specializing in Computer Science with a focus on Artificial Intelligence and Machine Learning at Pranveer Singh Institute of Technology, Kanpur, India. I have hands-on experience with Python, C, and C++, and have actively worked on projects involving software development, backend engineering, and database management.

I am very enthusiastic about contributing to the BRL-CAD Python Bindings project as part of Google Summer of Code 2025. I would greatly appreciate your guidance on how I can increase my chances of being selected for this project.

Would making early contributions before the proposal submission be beneficial? Is there a code challenge or task that I should complete to better demonstrate my interest and skills? Additionally, are there any specific areas in the project where contributions would be particularly impactful or appreciated at this stage?

I am eager to learn and contribute meaningfully to the BRL-CAD project, and I look forward to your advice on how to proceed effectively.

Thank you for your time and consideration.

Best regards,
Tanushree Sarkar
B.Tech CSE (AI & ML), 3rd Year
Pranveer Singh Institute of Technology, Kanpur, India

drossberg, 2025-04-09:
Hi @TanushreeSarkar, thank you for your interest on BRL-CAD. I'm afraid that it's too late for this years GSoC. The proposal submission deadline was yesterday.

However, you can still join our Zulip chat, ask questions, and familiarize yourself with BRL-CAD.