opencax/GSoC #53: Integrated language help feature

t-paul, 2022-02-07:
## Outline

Add more interactive help features for built-in functions and modules. Right now there's already a nice summary of parameters linked as [cheat sheet](http://www.openscad.org/cheatsheet/). Scope of this projects would be to use this information in extended form and make it available in a more direct way in the editor.

## Details

* Convert the cheat sheet information into machine readable format
* Find a way to generate the existing HTML format based on the core data
* Add context help to editor giving help for built-in functions and modules, e.g. by adding formatted help output to the console window, including the links to further documentation like the language manual on Wikibooks

## Expected Outcome

Cheat sheet is integrated into the application and additional context help for built-in functions and modules is available.

## Project Properties

### Skills

* Programming language is C++
* GUI programming with the Qt framework

### Difficulty 

Easy

### Size 

Medium (175h)

### Additional Information

* Potential mentor(s): [Marius Kintel](https://github.com/kintel) ([IRC](https://web.libera.chat/?channel=#openscad): kintel), [Torsten Paul](https://github.com/t-paul/) ([IRC](https://web.libera.chat/?channel=#openscad): teepee)
* Organization website: https://www.openscad.org/

qwfrankcheng, 2022-04-04:
Hello mentors! My name is Frank Cheng, a CS student at Cornell Univ. Sorry that I learned about GSoC late and just started out. I'm new to open source and excited to contribute to this project idea.

I currently have the OpenSCAD application and source code on my system. I would be grateful if you could give me some guidance on getting started on this project. 

Additionally, for the project idea Details, what does the "core data" refer to? ("Find a way to generate the existing HTML format based on the core data")

Thank you!

t-paul, 2022-04-04:
Right now there's some initial information embedded in the source which is mainly used in the editor for the popup hints.

https://github.com/openscad/openscad/blob/bd78e4162ac14172087c5be7aaf29372b7093035/src/core/primitives.cc#L946-L951

This could be extended and used in other places and contexts (e.g. more detailed help in editor, help display in console windows, ...).

In addition the data could be used to generate and show the [cheat-sheet](http://openscad.org/cheatsheet/snapshot.html) in the application instead of the external link.

For a more interactive discussion, find us on [IRC](https://web.libera.chat/?channel=#openscad).

purviporwal0, 2023-01-25:
I'd want to work on this issue. Please give this problem to me.

linktorahulraj, 2023-03-19:
@t-paul   I find the project interesting, I want to work on this issue. Please tell me where to start.

t-paul, 2023-03-19:
@linktorahulraj First step would be setting up a development environment for OpenSCAD and getting and building the latest code. Depending on what OS you are using this is a bit different but should follow the documentation. Probably easiest setup is on Linux but macOS and Windows should work find too.

linktorahulraj, 2023-03-22:
@t-paul I setup the  OpenSCAD environment and open this link given above by you https://github.com/openscad/openscad/blob/bd78e4162ac14172087c5be7aaf29372b7093035/src/core/primitives.cc#L946-L951
 what next thing do this please elaborate

linktorahulraj, 2023-03-22:
@t-paul When I ran the code on OpenSCAD environment the its showing 
Parsing design (AST generation)...
[ERROR: Parser error: syntax error in file openscadfile1.scad, line 27](https://github.com/opencax/GSoC/issues/27,G:/openscadfile1.scad) 
Execution aborted

linktorahulraj, 2023-03-24:
@t-paul Please provide me some resourses to solve the issue.

t-paul, 2023-03-24:
@linktorahulraj I don't see how I can help with the information you are giving.

You could start with the example files shipped with OpenSCAD and have a look at the tutorial for getting started. If you see an error, it's important to describe exactly what you are doing and giving the context needed to reproduce the problem, e.g. by providing the file causing the error (assuming it's small or trimmed down to focus on the error).

For some general tips on how to approach this, have a look at the [guideline page](https://opencax.github.io/gsoc_guidelines.html).

ahmed-kamel646, 2024-02-15:
Hello mentors! My name is Ahmed Mohamed Kamel , a CS student ,and I am excited to express my interest in participating in this project .
Can you give me a hint to get started?

awschult002, 2024-07-01:
I am not a student, nor am I involved in the summer of code. I am just a random developer that has stumbled onto OpenSCAD. 

For documentation, I would highly recommend checking out [Doxygen](https://www.doxygen.nl/) which should be able to create HTML and LaTeX output from source code. Using a simple custom command notation could help to sort out what function documentation should go into a public API help document or not.

Also, I would highly recommend the integration of the [Language Server Protocol](https://microsoft.github.io/language-server-protocol/). This will provide real-time help popups in the editor, and if the project space is setup correctly, then it will also provide real-time help pop ups for any function that is written within the current session. This is, not only, super powerful; but platform agnostic. So if someone wanted to code using Vim or Emacs, they could still get help popups by accessing the LSP.

t-paul, 2024-07-01:
@awschult002 this is the wrong place for general discussion, the topics you mention are already covered in the [main repository](https://github.com/openscad/openscad/) but are out of scope for GSoC. In fact there's even a prototype for LSP.

shashwatvoa, 2024-12-19:
Hello mentor sir i would like to contribute in this code base so can you help me out with the initial steps

t-paul, 2024-12-22:
@shashwatvoa the steps described earlier in this discussion here still apply, if you are interested, following those should get you started. Note that GSoC 2025 is not yet officially announced, according to some info from the mailing list, it's likely to happen very early in 2025.

closed 2025-02-09 by t-paul