opencax/GSoC #54: Font Selector Window with preview and search functionality

t-paul, 2022-02-07:
## Outline

OpenSCAD supports extracting geometry of text based on fonts installed on the system. The existing font list gives an overview of what fonts are installed, but does not provide further help to select fonts. An improved selection window could improve the user experience by giving the user a preview of some sample text and more ways for selecting a font.

## Details

* The current font list is implemented as dialog, this should be changed to a dockable window similar to the console
* Preview area should allow the user to enter some (limited) text which is then rendered in either a selectable size or maybe in multiple sizes
* Allow font selection by character, this will allow the user to find fonts that support specific unicode codepoints (e.g. for emoji)
* Select by font name + drag & drop to the editor - keep the feature of the existing font dialog

## Expected Outcome

The font selection dialog is replaced with a dockable font selection window with extra functionality.

## Project Properties

### Skills

* Programming language is C++
* GUI programming with the Qt framework

### Difficulty 

Medium

### Size 

Long (350h)

### Additional Information

* Potential mentor(s): [Marius Kintel](https://github.com/kintel) ([IRC](https://web.libera.chat/?channel=#openscad): kintel), [Torsten Paul](https://github.com/t-paul/) ([IRC](https://web.libera.chat/?channel=#openscad): teepee)
* Organization website: https://www.openscad.org/

UncleWeeds, 2024-02-25:
@t-paul, Interested in this project, would like to get more info on this

MJ2021, 2024-03-05:
Hi, @t-paul @kintel @UncleWeeds I am interested in this project could you provide some more information regarding this ?
Thanks

t-paul, 2024-03-05:
Have a look at the [Checklist](https://opencax.github.io/gsoc_checklist.html), it gives starting points. The most important ones being getting setup with the development environment and joining the IRC channel for discussion (consider that people are in different time zones so sometimes it takes a while for responses).

closed 2025-02-09 by t-paul