opencax/GSoC #57: Simple Library Manager / Downloader

t-paul, 2022-02-10:
## Outline

People have written a large number of libraries for use with OpenSCAD over the years. A list of some better known ones can be found in the [library](https://openscad.org/libraries.html) list on the homepage. So far it's difficult for new users to find libraries and the process of integrating them into a project is all manual work.

The main idea of this project is to provide some simpler means of discovering and installing libraries. One example of how this could be done can be found in the Arduino IDEs [Library Manager](https://docs.arduino.cc/software/ide-v1/tutorials/installing-libraries).

## Details

* Check #3479 for more ideas and references
* Define how the dependency specification will look like
* Support a separate lock file
* Limit to some provider (github) but be open for multiple source
* Needs to be usable from both command line and GUI eventually, scope of this project limited to command line only
* Implementation as library so it can be used in different contexts

Main process:
1) Download index file from a fixed location
2) Check dependencies against index file
3) Download library versions if needed (not in cache)

## Expected Outcome

Users can declare library dependencies and those will be downloaded, installed and made available for OpenSCAD scripts.

## Project Properties

### Skills

* Programming language is C++
* Knowledge of network programming

### Difficulty 

Medium

### Size 

Medium (175h)

### Additional Information

* Potential mentor(s): [Marius Kintel](https://github.com/kintel) ([IRC](https://web.libera.chat/?channel=#openscad): kintel), [Torsten Paul](https://github.com/t-paul/) ([IRC](https://web.libera.chat/?channel=#openscad): teepee)
* Organization website: https://www.openscad.org/

References:
* [Package manager: Discussion](https://github.com/openscad/openscad/issues/3479)

t-paul, 2022-03-16:
Yes, sure! I'd recommend discussing the details on IRC as the GSoC proposal will require a good overview of the technical approach and a development timeline.

paul-111, 2022-03-16:
@t-paul  hii i dropped messege in the Zulip and getting no response there

brlcad, 2022-03-16:
@paul-111 responses on any chat forum are often not instantaneous (you didn't even wait 20 minutes before declaring "no response", people multitask), but Zulip is also for BRL-CAD not OpenSCAD, so you'll want to join the OpenSCAD IRC as linked from https://opencax.github.io

t-paul, 2022-03-16:
You can also just click on the `IRC` links in the issue description. Those go to the IRC web frontend where you can join the `#openscad` channel even without the need to register.

rvt, 2022-03-16:
Can we do something like this?
```
library  <https://github.com/revarbat/BOSL2>
include <BOSL2/walls.scad>
```

Some thoughts...
library could handle various tags, hashes or branches. Perhaps the library should include a form of package/library file so openscad knows how to install it so include can find it.

Also an direct include would be nice `include <https://github.com/revarbat/BOSL2/walls.scad>`

The idea is still that when you download a scad file, it includes everything instead of manually installing files through a GUI...

t-paul, 2022-03-16:
No, at this point there's no plan to make OpenSCAD processing actively do network access.

rvt, 2022-03-16:
I was hoping for a system that would more look like platformio so each openscad project would have it's own version of libraries.

At least when I spoke to people about the Arduino system is that's it's a hassle to have different projects use different libraries and the versions... Perhaps that's changed, I have not use Arduino for a while.

Not trying to convince, but I think also scad files should be project based and with Arduino that is not really the case...

t-paul, 2022-03-16:
That's all good for discussion. If platformio logic would work for OpenSCAD, that's certainly a possible option too.

That said, there's right now *no* solution. Getting something useful into the hands of people is probably better than waiting 20 more years hoping for the perfect solution.
I had some hopes for entropic which was claiming to become a package manager for "all the things" but it looks abandoned now. vcpkg-ce is something that could be made to work probably, but GUI integration might be problematic and it's importing a whole new runtime environment which is not ideal either.

closed 2025-02-09 by t-paul