opencax/GSoC #58: LibreCAD 3 GUI

feragon, 2022-02-12:
## Outline
Improve the current GUI or implement a completely new one.

## Details
LibreCAD 3 has a GUI made with Qt and Lua. It is currently not very user-friendly and some improvements should be done in this direction.

The userbase is very broad and especially includes inexperimented users with LibreCAD. They should get started as quickly as possible.

## Expected Outcome
An improved GUI and possibly some mock-ups for functionnalities that can't be implemented 

## Future Possibilities 
- themes support

## Project Properties

### Skills
- C++ / Qt
- Lua
- GUI design

### Difficulty 
Medium

### Size 
medium (175h)

### Additional Information
* Potential mentor(s): [Florian](https://github.com/feragon)
- Organization website: [https://librecad.org/](https://librecad.org/)
- Communication channels: [https://librecad.zulipchat.com/](https://librecad.zulipchat.com/)

ReejoJoseph1244, 2022-03-16:
Greetings,
I am interested in contributing to this project as a part of GSOC 2022. I'm beginner to Open source . I have knowledge in C/C++,Python . can u guide regarding this project and how to start. I dont have much knowldege in Qt and Lua , but looking forward to start those .

It would be great if the mentors could guide me towards contributing and proposal

Thank You

lordofbikes, 2022-03-21:
Hey @ReejoJoseph1244, any issues finding our Zulip chat?
Is anything vague? Anything we can do for you?

closed 2023-03-05 by lordofbikes