opencax/GSoC #59: LibreCAD 3 Distribution

feragon, 2022-02-12:
## Outline
 Create distributions of LibreCAD 3 for different operating systems.

## Details
Users who want to try or use LibreCAD 3 have to compile it. This is too difficult for non-technical users. 

The installation procedure should be as simple as :
- Download the distribution file on LibreCAD website or GitHub releases
- Install the downloaded file
- Run LibreCAD

The packaging method is not imposed. The users have already requested the following :
- [Snap](https://snapcraft.io/)
- [AppImage](https://appimage.org/)

## Expected Outcome
Procedure or script to create a working distrubtion file of LibreCAD for Windows and Linux.
The distribution files will be tested on fresh installs of major Linux and Windows installations.

macOS builds are not required but it would be a plus.

## Future Possibilities
- Build and release those files automatically on GitHub

## Project Properties

### Skills
- Operating Systems
- C++

### Difficulty 
medium

### Size 
medium (175h)

### Additional Information
* Potential mentor(s): [Florian](https://github.com/feragon)
- Organization website: [https://librecad.org/](https://librecad.org/)
- Communication channels: [https://librecad.zulipchat.com/](https://librecad.zulipchat.com/)

CRiSTiK24, 2022-03-25:
Hi, second year university student from UdG, Spain here and I'm interested on this. The objective of the project is to automate [this](https://github.com/LibreCAD/LibreCAD_3/wiki/Building) installation right? If so, i'll try to install it myself on a linux and windows to check it out.
Should I communicate through here or in [https://librecad.zulipchat.com/](https://librecad.zulipchat.com/#narrow/stream/112841-Google-Summer-of-Code/topic/2022) or email/anything else? In [GSoC Participation Expectations](https://opencax.github.io/gsoc_expectations.html) it says to do it in IRC or Mailing list, Is librecad.zulipchat.com the IRC?

lordofbikes, 2022-03-26:
Welcome @CRiSTiK24!

We at LibreCAD prefer Zulip chat, which is linked in the task definition above.
You can follow the link and use a browser to log in or you can install their desktop client which can be found at https://zulip.com/apps.
There is already a topic [2022](https://librecad.zulipchat.com/#narrow/stream/112841-Google-Summer-of-Code/topic/2022) in the `Google Summer of Code` stream for this year.

The goal of this task is not only to automate the build process. It is also about building binary packages for Windows, Linux and MacOS, which could be downloaded by users for an easier installation.
So being able to build LibreCAD 3 on different OS is the first step.
Next is to explore GitHub Actions, because this should be the base to automate everything.
Thus this task will also need to set up the action runners to build LibreCAD 3 and to create and upload the binary packages from there.

We already do so for [LibreCAD 2](https://github.com/LibreCAD/LibreCAD/releases/tag/latest), but the requirements and build process for LibreCAD 3 are totally different. And we have no packaging at all for LibreCAD 3 yet.

closed 2023-03-05 by lordofbikes