opencax/GSoC #64: appleseed physically-based renderer release infrastructure

brlcad, 2022-02-25:
## Outline

Update [appleseed](https://appleseedhq.net)'s CMake build infrastructure, CI/CD scripting, and plugins.  Generate new up-to-date releases automatically.
 
## Details

The [appleseed physically-based renderer](https://appleseedhq.net) needs a variety of infrastructure updates made to its CMake build system and CI/CD release infrastructure so that updated release binaries are created and published automatically.  Appleseed leverages a number of third-party dependencies such as Ilmbase, Python, and OSL, and appleseed's CMake build logic and source code requires changes to support the latest versions of all of them.  Updating the embedded Python from version 2 to version 3 will require specific attention.

This project also includes setting up build and integration testing so that appleseed's integrations (e.g., Maya plugin, Blender plugin, etc) are also updated automatically.  Plugin and regular appleseed release builds need to be integrated with existing CI/CD shell scripting logic and GitHub Actions integration.

## Expected Outcome

Expected outcomes are 1) updates to appleseed's build logic and source code to utilize the latest of all current dependencies, 2) updates via GitHub Actions and/or appleseed's other CI/CD infrastructure

## Future Possibilities 

Make your own suggestions...

## Project Properties

### Skills

CMake, C/C++, Python, Shell scripting

### Difficulty 

Easy

### Size 

long (350h)

### Additional Information

* Potential mentor(s): [Sean Morrison](https://brlcad.org)
* Organization website: [appleseed physically-based renderer](https://appleseedhq.net)
* Communication channels: https://brlcad.zulipchat.com

linktorahulraj, 2023-03-19:
@brlcad I want to work on this issue, please tell me where to start

joe2911, 2023-03-20:
I'm an Engineering student with minimal experience using the listed languages. I'd like to participate if possible

mavavilj, 2023-07-04:
Is this still uncompleted?