opencax/GSoC #67: Continuation of a prior BRL-CAD GSoC effort

brlcad, 2022-03-09:
## Outline

BRL-CAD has been participating in GSoC for over 10 years with nearly 100 students!  Any past accepted projects can be submitted as a continuation project.
  
## Details

You can find all past participants [documented on BRL-CAD's wiki](https://brlcad.org/wiki/Google_Summer_of_Code) by selecting a given year (e.g., [2018](https://brlcad.org/wiki/Google_Summer_of_Code/2018)).  Even the most successful and completely integrated projects have room for improvement!  If any of those past efforts for any prior year sound very interesting to you, you can propose a continuation effort for it.

Of course, you will need to research the prior effort to determine the status of the work, whether code was integrated or is sitting pending integration in a patch, whether it's functional or was in an intermediate state, etc.  You'll also want to come [chat with us on Zulip](https://brlcad.zulipchat.com) to make sure there is mentoring support for it, but there usually is if you're passionate and independently productive.

For your proposal, note that it's a continuation effort.  Explain what you are doing and how it relates to the prior effort.  It's strongly recommended that your development plan focus on production-quality integration aspects such as making sure there are no usability or user experience (UX) issues, no build integration issues, that testing is covered adequately, and with focus on UX.

## Expected Outcome

The expected outcome of a continuation effort is new capability and features that are "complete", integrated, bug-free, and issue-free, in the hands of users.  This means your project covers all vertical integration aspects of development integration including build system and usability / UX concerns.  Not prototyped.  Not simply rewritten or re-attempted.

If the prior effort was integrated, your outcome will be specific polish, adaptiveness, and robustness improvements.

If the prior effort was not integrated, your outcome will be issue-free integration that addresses prior issues preventing integration (which will require research and understanding on your part).

## Project Properties

### Skills

This varies greatly by continuation.  There are continuation projects for C/C++, Python, Javascript/Node.js, Tcl/Tk, OpenCL, OpenGL, Qt, GPGPU, and more.

### Difficulty 

Varies.

### Size 

You are welcome to scope your project medium (175h) or long (350h) depending on the objectives and development scope.

### Additional Information

* Potential mentor(s): Morrison (contact [devs@brlcad.org](mailto:devs@brlcad.org))
* Organization website: https://brlcad.org
* Communication channels: https://brlcad.zulipchat.com

Trashman247, 2022-04-11:
Hello,
Incoming college freshman in comp sci - just learning about GSoC. I have coding experience in HTML/CSS + Python and am passionate about 3D design w/ Blender. This project seems to play to a wide range of skill levels and interests, so I hope finding the right one will challenge me just right. I would love to work with close cooperation with mentor(s) along the way as I learn! Let me know how I can start!

Thanks!

aothms, 2022-04-11:
Welcome. You can use the labels to filter based on your interests, for example there is quite a bit for to be done in Python https://github.com/opencax/GSoC/labels/Lang%3A%20Python

All the subprojects here have their own communication channels, so once you have a general idea what projects you'd like to contribute to, reach out in the appropriate channel.

Note that in this organization we have the strong recommendation to also try and get some code commits in along with the proposal to also familiarize yourself on that front.