opencax/GSoC #72: Interface to interact with Brickschema

Moult, 2023-02-26:
## Outline

Brickschema is an open data standard used to describe building services and systems, which may be used in Building Management and Control Systems (BMCS). For example, it may describe many sensors, equipment, and real-time statistics on sensed data, like temperature, airflow, etc.

Your job will be to update the existing module in the BlenderBIM Add-on to use the latest Brickschema library, add undo/redo support, as well as add basic features for adding / removing / editing sensor points, timeseries data, and more.

## Expected Outcome

The user will be able to edit Brickschema models in the interface.

### Skills

 - Python
 - You can learn about [Brickschema](https://brickschema.org/) on the project, it is an RDF based data schema describing buildings

### Difficulty 

Easy

### Size 

 - **Medium**: Brickschema interface is updated, test cases written, undo/redo support added, basic ability to add / remove sensor points.
 - **Long** Additional ability to add / remove feed relationships (e.g. this equipment is feeding another equipment), manage XML namespaces, and possible IFC conversion.

### Additional Information

* Potential mentor(s): Dion Moult
* Organization website: https://ifcopenshell.org
* Communication channels: https://community.osarch.org , ##architect on Freenode , and https://github.com/IfcOpenShell/IfcOpenShell/issues

Some screenshots of the existing bare-bones Brickschema interface are shown:

[image]
[image]
[image]

rileywong311, 2023-02-27:
Hi IfcOpenShell! I am interested in learning more about this project (and your other "gITF compatible materials" project). I have some experience with the Blender API already, but would need to learn more about Brickschema and how editing the data that comes with that works. I am a little confused about where I can stay in communication, so let me know!

Moult, 2023-02-27:
G'day @rileywong311 ! The best way to chat is in the osarch.org/chat chatroom (via IRC or Matrix). It would be great to learn more and discuss your application!

senyasuper, 2023-03-29:
Hello Dion, if this topic is possible to split, I'd like to participate. Told a bit about myself in osarch chat

Moult, 2024-01-07:
Closing as this has now been implemented.

closed 2024-01-07 by Moult