opencax/GSoC #73: Interface to easily create glTF compatible materials for IFC objects

Moult, 2023-02-26:
## Outline

3D building models typically have very simple materials. It is possible for them to have complex materials and textures, compatible with the glTF standard. Right now, there is no interface to allow users to easily create these node setups and visualise these beautiful textures.

Your job is to design and create an interface in Blender to allow users to quickly select image textures, bump maps, shading styles from drop downs, and so on. This would then apply the relevant materials to the selected objects.

### Skills

 - Python
 - Knowledge of Blender materials

### Difficulty 

Easy

### Size 

Medium

### Additional Information

* Potential mentor(s): Dion Moult
* Organization website: https://ifcopenshell.org
* Communication channels: https://community.osarch.org , ##architect on Freenode , and https://github.com/IfcOpenShell/IfcOpenShell/issues

[image]

thejastm15, 2023-03-03:
Hey i am intrested in this project for GSoC

brlcad, 2023-03-03:
There are lots of ways to go about this project, but might be useful reference to check out the [Blenderseed](https://github.com/appleseedhq/blenderseed) plugin that leverages the nodal editor in Blender to specify complex materials and textures for use with the Appleseed renderer.  This is very similar if not identical to what is needed for exporting glTF-compatible material shader networks.  The Blenderseed code could be adapted to export both, or simply used as a code reference for understanding how to make a Blender plugin.  It has nice docs too:  https://appleseed.readthedocs.io/projects/appleseed-blenderseed/en/2.0.0-beta/

brlcad, 2023-03-03:
This is also potentially a project of mutual interest to BRL-CAD as it received support for glTF and OSL shaders (via appleseed rendering) just this past year.  Happy to support any collaborative goals.

Moult, 2023-03-04:
@thejastm15 great! Essentially there are a few ways to set up material nodes in Blender that are considered to be "gltf compatible".

However setting up these nodes requires you to know the node setups https://docs.blender.org/manual/en/dev/addons/import_export/scene_gltf2.html

It would be good to have an interface to allow users to one click add and change material settings, and the node setups are always correct.

In addition, doing things like one click add image plane (in a gltf compatible manner) is very useful for AEC, like for modeling signage.

thejastm15, 2023-03-05:
@Moult how to connect with the commnity

Moult, 2023-03-05:
@thejastm15 the best place is osarch.org/chat via either IRC or Matrix. See you there!

Yuvraj9394, 2024-01-03:
@brlcad Hello I am interested in that project for GSoC 2024. I have intermediate knowledge of blender and python and i want to learn more by contributing and  participating in this project.

Moult, 2024-01-07:
Sorry I think this has now been implemented.

closed 2024-01-07 by Moult