opencax/GSoC #76: KiCad to ODB++ exporter

sethhillbrand, 2023-03-03:
Outline
-------

KiCad currently supports export to GenCAD, Gerber, Gerber X2 and IPC-D-356. These are commonly used, but many manufacturers would like the additional information contained in an ODB++ file.  This includes net information, board stackup, bill of materials, embedded stroke fonts as well as standard layer data.

Details
-------

The ODB++ Design website provides a reference viewer to verify data output. There are many examples in the existing Gerber and GenCAD exporters to use for implementation frameworks.

Relevant Links:

- https://odbplusplus.com/design
- https://gitlab.com/kicad/code/kicad/-/issues/2019

Expected Outcome
----------------

An ODB++ exporter that can generate standard ODB++ files using existing v8 KiCad features.

Project Properties
------------------

### Skills

- C++
- Basic knowledge of EDA software concepts

### Difficulty

Medium

### Size

350 hours

### Additional Information

- Potential mentors:
    - [Seth Hillbrand](mailto:gsoc@kicad.org)
- Organization website: https://kicad.org
- Communication channels:
    - gsoc@kicad.org
    - [Discord channel](https://discord.gg/WhzyWZ5RaD)

Neeraj8949, 2023-03-04:
I want to contribute in this software .I have good knowledge of c++ and for this I am going through EDM software to proceed this issue

sethhillbrand, 2023-03-13:
Hello @Neeraj8949 We're happy that you are interested in joining KiCad for the GSoC!  The first step will be to introduce yourself over on our Discord channel.  We have some intro steps that you will want to look at.

UncleWeeds, 2024-02-25:
@sethhillbrand Hey my self Abhiram, I am participating in gsoc'24. Really interested in this project, any advice where to get started would be really helpful

sethhillbrand, 2024-02-26:
Hello @UncleWeeds .  This issue is out of date.  We should be merging ODB++ this year.  KiCad is not part of gsoc this year.

closed 2024-02-26 by sethhillbrand