opencax/GSoC #77: Create a new KiCad Importer

sethhillbrand, 2023-03-03:
Outline
-------

Create a new importer for files from any EDA software for which there is no built-in importer yet.

Details
-------

A non-exhaustive list of possible EDA software:

- Easy (open source): Fritzing, gEDA, Horizon EDA, LibrePCB, JSchem, pcb-rnd
- Easy, but slightly harder (port existing importer): EasyEDA ([link](https://github.com/wokwi/easyeda2kicad), [link](https://github.com/RigoLigoRLC/LC2KiCad)), Zuken CR-5000 ([link](https://forum.kicad.info/t/zuken-cr-5000-to-kicad-converter/22923))
- Medium (text formats): Allegro ASCII, PADS ASCII, Mentor HyperLynx, SIMetrix, Micro-Cap, ...
- Hard (binary formats): Target 3001!, DipTrace, EAGLE < 5.91, Allegro, Mentor, ...

There are two FOSDEM talks around this topic to get a rough idea of the task:

- [Importing into KiCad from CADSTAR](https://archive.fosdem.org/2021/schedule/event/openhard_cadstar/)
- [Reverse-Engineering of (binary) File-Formats](https://archive.fosdem.org/2021/schedule/event/reverse_engineering/)

In case you want to write an importer for a proprietary EDA program, you need to have legal access to this software. You can use a demo or student version if it is available, or you may have some other means of access, like your university institute (please ask them beforehand). If you are unsure about this topic, choose one of the open-source tools instead.

Expected Outcome
----------------

An importer for files from EDA software for which there was no built-in importer. Depending on the complexity, this might include board, schematic and/or library import.

Project Properties
------------------

### Skills

- C++
- Basic knowledge of EDA software

### Difficulty

Depends on the selected EDA package.

### Size

175 hours (Open source or porting existing importer)
350 hours (Create new importer from scratch--documentation or reference implementations are available)

### Additional Information

- Potential mentors:
    - [Seth Hillbrand](mailto:gsoc@kicad.org)
- Organization website: https://kicad.org
- Communication channels:
    - gsoc@kicad.org
    - [Discord channel](https://discord.gg/WhzyWZ5RaD)

Neeraj8949, 2023-03-04:
I want to contribute in this issue .Can you help me to how I proceed

sethhillbrand, 2023-03-13:
The source of KiCad is located at GitLab.  We have a detailed writeup on how to download and compile the KiCad sourcecode at https://dev-docs.kicad.org/en/

skushagra9, 2023-12-21:
Hey , I am here for Gsoc , Please guide me how should I proceed

closed 2024-04-10 by sethhillbrand