opencax/GSoC #81: Improve mesh ID handling

pca006132, 2024-01-21:
## Outline

Improve mesh operation performance by making mesh ID tracking opt-in.

## Details

We currently track mesh id for boolean operations and compose, which allows users to figure out where a particular triangle comes from and reapply certain properties. Triangles from different meshes will not be collapsed, so in some cases tracking this information can have significant overhead (https://github.com/elalish/manifold/pull/671).

We can make mesh ID tracking an opt-in feature.

- Default mesh ID is 0.
- `IncrementMeshIDs` will only remap non-zero IDs.
- We treat meshes with mesh ID 0 as from the same mesh, and coplanar triangles can be collapsed.
- To opt-out from this behavior, the user should call `AsOriginal` for the boolean operands to give the mesh some non-zero initial ID, and future boolean operations will give the result different mesh IDs due to calls to `IncrementMeshIDs`.
- Also, `AsOriginal` can accept an optional parameter to only update specific IDs.

```c++
// cubes have mesh ID 0, so result also have mesh ID 0
Manifold result = Cube() + Cube();
// give result2 mesh ID 1
Manifold result2 = result.AsOriginal();
// result3 has mesh ID 2 and 0, where ID 2 is from ID1 from result2, and 0 is from Cube.
Manifold result3 = result2 + Cube();
```

## Expected Outcome

Designed mesh ID behavior as above, and good performance.

## Project Properties

### Skills

- C++

### Difficulty 

- Easy

### Size 

- Medium (175h)

### Additional Information

* Potential mentor(s): @pca006132
* Organization website: https://manifoldcad.org
* Communication channels: https://github.com/elalish/manifold/discussions

MJ2021, 2024-03-05:
Hi @elalish for improving the existing algorithms are ideas already available or we need to come up with new ideas ?

pca006132, 2024-03-05:
the ideas are there, but the implementation may still need you to come up with some ideas to bridge the gaps.

pca006132, 2024-03-05:
you should just study the existing codebase. They are not independent, they should be done in parallel.

MJ2021, 2024-03-05:
Okay and how I should I start contributing now and prepare for my application ?

MJ2021, 2024-03-06:
Sure can you provide me the link that exactly where is this code, I am not able to find that 
Thanks

elalish, 2024-03-06:
No, this is good practice for you. Both VSCode and Github allow you to search an entire code base. See what you can find.

MJ2021, 2024-03-15:
Hi, sorry for the late reply I got a bit busy with some academic commitments. I have looked at the code base and I think the main changes that we need to do is in the "meshIO" folder where there are 2 files namely meshIO.h and meshIO.cpp, can you guide if I am going in the right direction ?

pca006132, 2024-03-15:
No, mesh simplification is run after boolean operation, not only when doing imports/exports.

MJ2021, 2024-03-27:
Hi, I have created a draft PR can we have more discussion there about the code ?
https://github.com/elalish/manifold/pull/779
Thanks

MJ2021, 2024-03-30:
Hi, can you provide some information how I should prepare my proposal for this project ?

pca006132, 2024-03-31:
I think the guideline is just the generic one, show your interest as well as past experience in courses/other work. It is fine that you was not able to do much in this project until now, but to be honest it will put you in slight disadvantage, there is no way for us to see your understanding on the subject, so you should show it in your proposal..

MJ2021, 2024-04-01:
Can you provide some suggestions on how I can improve my proposal and increase my chances of learning more https://docs.google.com/document/d/1nbpG2qp6sLyp4vVg3LaIYWKqhQTKWQMinds9pmI3qss/edit

MJ2021, 2024-04-16:
Is there an official discussion channel for the project ?

elalish, 2024-04-16:
This is it!

pca006132, 2025-01-26:
@elalish is already doing this.

closed 2025-01-26 by pca006132