opencax/GSoC #82: Add minGap function

pca006132, 2024-01-21:
## Outline

Implement `minGap` function to check mechanism tolerances. (https://github.com/elalish/manifold/issues/186)

## Details

A convenient function would be to measure the minimum gap between two Manifolds (good for checking that mechanisms have the proper tolerances). The signature would be: `float Manifold.MinGap(const Manifold& other, float searchLength) const`

It would always return a value between zero and `searchLength`. It would perform an intersection and if that was non-empty, it returns zero (overlapping). If not, it will collide the triangles of each mesh with the triangles of the other, expanded into boxes `2*searchLength` wide, then for those collisions, calculate triangle-to-triangle distances. The minimum is returned.

## Expected Outcome

Implementation of the `minGap` function.

## Future Possibilities 

An algorithm without needing for a `searchLength`. Perhaps it is possible by doing nearest neighbor search (to determine a good searchLength for the point) + range query (to get all potential nearest triangles).

## Project Properties

### Skills

- C++
- Basic understanding of graphs and algorithms.

### Difficulty 

- Medium

### Size 

- Medium (175h)

### Additional Information

* Potential mentor(s): @pca006132 @elalish 
* Organization website: https://manifoldcad.org
* Communication channels: https://github.com/elalish/manifold/discussions

mleleszi, 2024-03-04:
Hey @elalish @pca006132,

This seems like an interesting problem, I'd love to work on this! I've spent a bit of time going through the wiki and the codebase, got the build and tests working. I've put together a little POC based on the algorithm proposed in the project description to check if I'm understanding things correctly. (I'm just exploring things, disregard the code quality/organization)

https://github.com/elalish/manifold/compare/master...mleleszi:manifold:mingap-poc

Am I going in the right direction? Let me know what you think. I'd love to contribute even before GSoC starts!

mleleszi, 2024-03-04:
Thank you, I'll try these! How much of this should I ideally get done before GSoC? Get as much implemented and merged as I can, and If there isn't enough work left from this for GSoC then write a proposal for a different project idea? Or how should I got about it

elalish, 2024-03-04:
I'm very flexible; my impression of GSoC is that it's trying to encourage people to fall in love with open source and continue contributing. Certainly that's the best outcome for us. I will be happy to give you more project ideas or guide any you come up with to keep you as busy as you like. I'm sure we can make it work, so let's cross that bridge when we come to it.

mleleszi, 2024-03-04:
Got it, thank you!

mleleszi, 2024-03-06:
Added the test cases you recommended and it made me realize that the triangle to triangle distance calculation is a bit more tricky than I initially thought, I didn't cover a lot of cases. I've researched algorithms for it and opened a [draft PR](https://github.com/elalish/manifold/pull/765) where I documented it. We can continue the discussion there and it can serve as the plan/preparation for GSoC if I don't get to the implementation till then. I'll also look into your other recommendation as to how we could use the Collider to speed things up.

Moult, 2024-03-06:
Sorry just jumping in here not knowing any context but FYI Nvidia's Omniverse Physx has a whole bunch of algorithms available including triangle-triangle distance: https://github.com/NVIDIA-Omniverse/PhysX/blob/a2af52eb6a2532bd2bc583ef8ead9c81c9222af1/physx/source/geomutils/src/distance/GuDistanceTriangleTriangle.cpp

 - more distance algos: https://github.com/NVIDIA-Omniverse/PhysX/tree/a2af52eb6a2532bd2bc583ef8ead9c81c9222af1/physx/source/geomutils/src/distance
 - and of course intersection: https://github.com/NVIDIA-Omniverse/PhysX/tree/a2af52eb6a2532bd2bc583ef8ead9c81c9222af1/physx/source/geomutils/src/intersection

It's licensed under BSD. We use it in IfcOpenShell for collision stuff too. It sounds similar to what this project is describing: we have a BVH tree of every triangle mesh, collide BVHs and then use tri-tri distance to determine "clearance".

mleleszi, 2024-03-07:
Thank you @Moult, I'll take a look!

Moult, 2024-03-07:
If it helps, here is the IfcOpenShell implementation of a many-many clearance clash: https://github.com/IfcOpenShell/IfcOpenShell/blob/796760a064007bb63b9b330a9f7e2853c6ef9ff1/src/ifcgeom_schema_agnostic/IfcGeomTree.h#L1224

It uses a BVH which is very fast compared to looping through all triangles, and multithreading. Then the actual distance check is here: https://github.com/IfcOpenShell/IfcOpenShell/blob/796760a064007bb63b9b330a9f7e2853c6ef9ff1/src/ifcgeom_schema_agnostic/IfcGeomTree.h#L791 with the modified tri-tri code (for OpenCASCADE) here: https://github.com/IfcOpenShell/IfcOpenShell/blob/796760a064007bb63b9b330a9f7e2853c6ef9ff1/src/ifcgeom_schema_agnostic/clash_utils.cpp#L192

Note that there are a number of probably bad coding practices in there since I'm a beginner to C++. We're slowly cleaning it up in https://github.com/IfcOpenShell/IfcOpenShell/tree/feature-clash3

mleleszi, 2024-03-07:
Thank you, these will definitely help!

elalish, 2024-03-07:
Thanks @Moult, though we actually already have our own parallel BHV tree (Coliider) that we use in our library. Still, the tri-to-tri stuff will likely be useful!

elalish, 2024-04-02:
Done, thanks to @mleleszi in https://github.com/elalish/manifold/pull/765!

closed 2024-04-02 by elalish