opencax/GSoC #83: Add fuzzing tests

pca006132, 2024-01-21:
## Outline

Add more fuzzing tests for both 2D and 3D operations.

## Details

[Fuzzing](https://en.wikipedia.org/wiki/Fuzzing) is an effective technique to expose bugs in software. Fuzzing tests randomly generate structured inputs (according to specification), and test if the program crashes/failed assertions.

This project aims to test 2D and 3D CSG operations on geometrically valid polygons/meshes. To do this, we will define a very simple AST for our CSG operations, and use the [recursive domain](https://github.com/google/fuzztest/blob/main/doc/domains-reference.md#recursive-domains) feature of [fuzztest](https://github.com/google/fuzztest/tree/main) for the tests.

We will also randomly apply slight perturbation to make the valid geometry only epsilon-valid, to test for robustness of the algorithm.

## Expected Outcome

Fuzz tests that test for union, intersection, difference, 2D extrude/revolve, etc.

## Project Properties

### Skills

- C++
- Basic understanding of graph data structure.

### Difficulty 

- Medium

### Size 

- Medium

### Additional Information

* Potential mentor(s): @pca006132 
* Organization website: https://manifoldcad.org/
* Communication channels: https://github.com/elalish/manifold/discussions

dasoya, 2024-03-20:
Hi, @pca006132  I am a senior college student in EWHA. I am interested in this project. I am new in using fuzz test but I implemented some graphics feature like ray tracing in OpenGL using C++. 

I looked up your fuzz test implementations for the triangulator here 
https://github.com/elalish/manifold/blob/master/test/polygon_fuzz.cpp

In this case, you've implemented tests for segfault/ memory corruption and infinite loops. I'm wondering if the CSG test is also designed to check whether the process of creating the AST can cause segmentation faults.

Additionally, I want to know if you've implemented any other fuzz test modules that I could use as a reference. I'm also interested in learning how CSG operations work. Thank you in advance for your suggestions.

pca006132, 2024-03-20:
Creating the AST should not cause segfault, but converting the AST to actual CSG operations, i.e. *interpreting* the AST, can cause segfault. And no, `polygon_fuzz.cpp` is the only fuzz test module we have for now, but feel free to ask us anything about the underlying CSG operation.

pca006132, 2025-01-26:
Changed to lazy because we only have to do csg tree fuzzing now.

moOsama76, 2025-02-27:
Hello, I am a junior engineer with experience in CAD and Graphics programmer 
I interned at fater.ai for few months which included making 3d models with Cadquery which is built over OpenCASCADE, and I contributed to graphite.rs which is a 2d art tool

https://github.com/GraphiteEditor/Graphite/pulls?q=author%3Amoosama76+state%3Amerged
https://github.com/CadQuery/cadquery-contrib/pulls?q=author%3Amoosama76+state%3Amerged

I am interested in this project because I'll be doing mandatory military service and can't propose to larger projects