opencax/GSoC #85: Scripts to generate light simulation input files from 3D models

Moult, 2024-01-21:
## Outline

The BlenderBIM Add-on and IfcOpenShell software allows users to create 3D models of buildings. Another software, [Radiance](https://www.radiance-online.org/), is used to perform light simulation (e.g. create renders of models that are photometrically correct). This project will integrate the two software together to create beautiful, detailed, but scientifically accurate 3D renders.

## Details

**No knowledge of light simulation is required!**

Light simulation software simply requires 3D models in obj format as well as a set of input text files that describe the materials and scene. You will be guided on the format of these simple text files, and what files need to be generated.

Other integration options include filtering what objects are converted for light simulation, creating a GUI in Blender (this is relatively straightfoward and can be learned during the GSoC project), object proxying (e.g. swapping out one object for another), and material swapping / mapping (e.g. specifying which objects have what materials).

## Expected Outcome

In order of priority (you may not complete all outcomes):

 1. Python scripts that take an input 3D IFC model (the model of the building) and produce OBJ files and text files that can be read by Radiance (light simulation software).
 2. A graphical interface in Blender to perform this conversion and provide settings.
 3. Settings that handle object filtering, object count limits, object geometry substitution, material mapping and/or substitution.
 4. Single plane substitution for glazing (glazing is typically modeled as a cuboid, but light simulation requires it to be a flat plane)
 5. Choose from preset sky settings and automatic materials

## Project Properties

### Skills

 - Python (required)
 - Knowledge of Blender (optional, but beneficial)
 - Knowledge of Radiance or 3D rendering software (not required at all, but if you are familiar with 3D rendering this will be a fun project for you to learn the depths and history of physical rendering)

### Difficulty 

Easy

### Size 

Medium or Long

### Additional Information

* Potential mentor(s): Dion Moult
* Organization website: https://blenderbim.org http://ifcopenshell.org
* Communication channels: https://osarch.org/chat

AnonymousCodes911, 2024-02-14:
Greetings!, @Moult 

I am intrigued by this project, and I have always wanted to learn more about 3D modeling,  I am proficient in Python and would love to be guided on what my first steps should be to dive deeper into this project and take me forward to my first Pull Request.

Thank you for your time and consideration

Moult, 2024-02-18:
G'day @AnonymousCodes911 ! GSoC 2024 has not yet started and we don't know yet if the organisation is approved. If you're keen anyway a good starting point would be to install Radiance and do one of their beginner tutorials or perhaps this one: https://thinkmoult.com/basic-rendering-tutorial-radiance.html

akshayb80, 2024-02-24:
Greetings @Moult!

I’m a Master’s student in CS and very enthusiastic about contributing to this project. I have a strong interest in 3D modeling and prior experience with Blender. Additionally, having used Python extensively in my software projects, I believe I can make meaningful contributions to this project.

I would appreciate your guidance on the initial steps to deepen my involvement leading to submitting my first pull request.

Thank you!

DanielaHz, 2024-02-26:
Hello @Moult :)
 I'm Daniela, and I'm really interested in this project, and the #86. After analyzing the project details, I have a few questions I hope you can clarify:

1. I notice the project aims to use Radiance for light simulation. Could you please explain the reasoning behind choosing this software? Is it due to any specific restrictions, preferences, or perhaps because of a successful implementation seen in similar software?"
2. Regarding the graphical user interface in Blender, do you have any specific references or design preferences in mind? Additionally, concerning the settings those are the mentioned in the third outcome list, i mean: object filtering, count limits...?
3.  Could you provide guidance on the format of the required text files and specify the types of files that need to be generated for the integration?"
4. Regarding the proposal for GSoC, could you please clarify the format? Is it typically a repository on Git, a Word document, or another format?
5.  While I understand that completing all the outcomes listed in the project description may not be expected, would creating a proposal to integrate all these outcomes in actionable steps potentially enhance my evaluation?
Thanks

Moult, 2024-02-27:
Hey @AnonymousCodes911 @akshayb80  The organisation has been approved so now is an excellent time to make a start. The checklist is here: https://opencax.github.io/gsoc_checklist.html and you can reach out at https://osarch.org/chat

Apart from the initial steps to install Radiance and do the beginners tutorial as described above, the checklist will help. Say hello in the community chat, start writing an application, and play around with software! The best way to write a strong application is to tinker with the tech :)

@DanielaHz Radiance is the gold standard for light simulation, backed by many decades of rigorous scientific validation, and is also free software. The graphical UI in Blender should integrate into the existing UI that the BlenderBIM Add-on provides. The format of the text files will be clear once you start playing around with the Radiance software and doing the beginner tutorials. The GSoC project proposal can be any format. Guidelines on what should be on the application is in the checklist link above. A strong application is one that shows 1) you know what you're talking about 2) you've put in an effort to play with the software and see what the task entails and where the challenges would be, and 3) you're participating with the community :)

Royalty252, 2024-03-03:
Hello @Moult ,

I'm Avadhut and I'm keenly interested in your project. With my proficiency in Python and prior experience using Blender for animations, I'm particularly drawn to the idea of developing Python scripts to streamline conversions and creating a user-friendly interface in Blender for customizable settings.

I'm eager to contribute to this project and would appreciate any guidance on how to formally propose my ideas. Looking forward to your thoughts!

Thank You!!

Moult, 2024-03-03:
@Royalty252 g'day! Start by looking at the checklist: https://opencax.github.io/gsoc_checklist.html and you can reach out at https://osarch.org/chat - when you've got an idea of how to proceed, start drafting an application proposal and reach out for a time to go through it 1 on 1 in the chatroom :)

chiragsingh1711, 2024-03-08:
Hello @Moult ,
I am Chirag and I am writing to express my interest in contributing to the integration project between BlenderBIM and Radiance for light simulation. I have been actively involved in using Blender for modelling and scripting for quite a few years, specializing in automating tasks and implementing various automation scripts. While I do not have direct experience with Radiance, I am eager to learn and contribute my skills to this exciting project.

I am excited about the prospect of contributing to this project and believe that my skills align well with the outlined tasks and would love getting in touch with you and discuss more about the project.

Best regards,
Chirag Singh

Moult, 2024-03-08:
G'day @chiragsingh1711 ! It would be awesome if you could work on this project! Be sure to say hello in the OSArch chat, start following the beginner radiance rendering tutorial, and start drafting a proposal. When you have something written down or ideas, let's have a voice call to discuss further!

Moult, 2025-03-06:
Closing as implemented.

closed 2025-03-06 by Moult