opencax/GSoC #86: Features for CG artists to visualise beautiful IFC models in Blender

Moult, 2024-01-21:
## Outline

The architecture, engineering, and construction industry creates 3D models of buildings. These models are generally quite poor and do not contain any textures, lighting, or high quality objects that are suitable for 3D rendering. They often hire artists to help create beautiful renders of their designs.

This project will build utility functions and workflows to easily get beautiful pictures of 3D models.

## Details

3D artists typically do the following steps to make a 3D model look beautiful. They:

 1. Set camera angles with specific camera settings, with "clay" (e.g. all white colours) materials.
 2. Add lights and sun / sky settings.
 3. Add simple colours and textures.
 4. Remodel low quality geometry
 5. Add new objects (e.g. entourage) to decorate the scene, like trees, grass, people, extra furniture, walruses, shrimp, etc.
 6. Set common compositing and post processing rules

You will use the Blender Python API to set simple presets for most of these steps to allow less skilled artists to quickly setup renders. You will also setup a workflow to guide artists on how to organise their files relative to the IFC model and keep the IFC model separate so that when the IFC model is changed, the artists doesn't need to start from scratch or play spot the difference.

**You do not need to be an expert in 3D modeling or CG visualisation or rendering**. You will be taught what type of settings and options are appropriate for presets and the details of the workflow. However, you will be expected to automate that detail (every aspect of the Blender settings can be set using Python trivially).

You will also be expected to create a Blender interface to interact with the settings, e.g. a button to add camera, a button to set a preset sky, etc.

## Expected Outcome

Note: scope is flexible and you may achieve less or more or different to the below:

 1. A graphical interface in Blender that relate to the 6 steps above
 2. Buttons to add cameras, set common camera aspect ratios and settings. Buttons to add common types of lights, set sun angles and sky settings with bundled HDRI textures.
 3. Buttons to add simple material presets.
 4. Buttons to mark an object to be replaced by another
 5. A few preset assets using Blender's built in asset tools to drag and drop in entourage.

## Future Possibilities 

## Project Properties

### Skills

 - Python (definitely required!)
 - Artistic sense (do you like 3D graphics? rendering?) If you have ever rendered a 3D scene before, this is the project for you!

### Difficulty 

Easy to Medium

### Size 

Medium to Long

### Additional Information

* Potential mentor(s): Dion Moult
* Organization website: https://blenderbim.org http://ifcopenshell.org
* Communication channels: https://osarch.org/chat

AnonymousCodes911, 2024-02-14:
Greetings, @Moult!

I am intrigued by this project, and I have always wanted to learn more about 3D modeling and rendering. I am proficient in Python and would love to be guided on what my first steps should be to dive deeper into this project and take me forward to my first Pull Request.

Thank you for your time and consideration

Moult, 2024-02-18:
G'day @AnonymousCodes911 ! GSoC 2024 has not yet started and we don't know yet if the organisation is approved. If you're keen anyway a good starting point would be to fire up Blender and get familiar with doing basic rendering (perhaps the doughnut tutorial?) to get a feel of the CG workflow.

Moult, 2024-02-27:
Hey @AnonymousCodes911 The organisation has been approved so now is an excellent time to make a start. The checklist is here: https://opencax.github.io/gsoc_checklist.html and you can reach out at https://osarch.org/chat

keshavanarayan, 2024-02-27:
Good day @Moult,

I'm proficient in 3d rendering and writing python scripts in other software. This seems like a good mix of my interests and also a nice way to get into blender and blenderBIM. I have been familiarising with ifcopenshell for a while and I think this is a good mix of my interests.

Do let me know if I can be of any help 

Cheers,
Keshava Narayan

keshavanarayan, 2024-03-01:
Is there any deadline for the proposal @Moult?

Moult, 2024-03-02:
https://developers.google.com/open-source/gsoc/timeline

The earlier you start on a draft and discuss it with us, the better :)

chiragsingh1711, 2024-03-08:
Hey @Moult ,

I'm eager to contribute to the BRL-CAX project with my extensive experience in **Blender** and **Python scripting**, particularly in automating 3D modeling tasks. Having tackled similar challenges before, I understand the industry's need for enhancing 3D models with textures, lighting, and high-quality objects for rendering.

I'm confident that my skills align seamlessly with the outlined project, and I'm eager to contribute to the creation of utility functions and **workflows** that simplify the rendering process for less experienced artists.

Best regards,
Chirag Singh

Moult, 2024-03-08:
@chiragsingh1711 g'day! That's fantastic, if you can start writing a draft proposal planning the necessary features, reach out on the chat and we can have a voice discussion about how to proceed :)

RuoyuLi97, 2024-03-09:
Good day @Moult ,

I am passionate about this project. My name is Ruoyu Li and I am an incoming grad student of Information Systems at Northeastern University this fall. I am proficient in python and as a former architect, I am also experienced in 3d modeling, rendering and art sense of course. I am wondering where I should start and I hope to hear from you. 

Thank you,
Ruoyu Li

Moult, 2024-03-10:
@RuoyuLi97 g'day! That would be awesome! Join the OSArch chat, say hi, start writing a draft proposal and reach out if you have ideas / questions. The earlier you reach out with tangible ideas the better :)