opencax/GSoC #88: Blender UI / integration with voxelisation toolkit software

Moult, 2024-01-22:
## Outline

There is software known as the [Voxelisation Toolkit](https://github.com/opensourceBIM/voxelization_toolkit). It takes an input text file with commands in it and parses a 3D model. It converts the 3D model into voxels (e.g. 3D cubes that represent geometry), analyses those voxels, and outputs statistics (e.g. distance between voxels, etc).

Voxels are super cool and can be used to calculate head heights, resolve complex non-manifold geometry, egress distances, or concrete formwork areas and strutting distances, and air volume for mechanical calculations. All of this stuff is useful for engineers and construction professionals.

This is currently only available to computer programmers who know how to code this file and write code to parse the results. Yikes! This project is to add a UI in Blender to start making this general purpose analysis tool available to non-programmers.

## Details

You will be expected to design an interface for the voxelisation toolkit, prepack some simple recipes, and write scripts that take the output (currently visualised as images or plots) and instead visualise the results in 3D by generating 3D coloured meshes that represent the output.

## Expected Outcome

 1. Bundle the voxelisation toolkit software with Blender.
 2. A UI to execute the voxelisation toolkit.
 3. Simple presets to run the toolkit.
 4. Visualise the output of the voxelisation analysis as a 3D coloured mesh.

## Future Possibilities 

Bundle scripts for common usecases, like formwork calculation, air volume calculation, or external / internal metadata addition.

## Project Properties

### Skills

 - Python

### Difficulty 

Medium

### Size 

Medium to Long

### Additional Information

* Potential mentor(s): Dion Moult, Thomas Krijnen
* Organization website: https://blenderbim.org http://ifcopenshell.org
* Communication channels: https://osarch.org/chat

dororo-ak, 2024-03-06:
Hello I'm Adama a computer graphic student and an art lover. This project is pretty fun ! If i well understand the goal is to create a blender add to provide a GUI for a voxel apps ?

Moult, 2024-03-06:
G'day @dororo-ak absolutely! Yes, the voxelisation toolkit is an existing app which you can play with (here's a sample IFC model you can use https://www.ifcwiki.org/images/e/e3/AC20-FZK-Haus.ifc ) and it needs a Blender UI to help non-programmers play with it. That Blender UI could be absolutely anything - a panel, interactive workspace tool, nodes, web-based, etc. It should integrate to some degree (at a minimum, be bundled with) https://blenderbim.org/

If you're up for it, start playing with the voxelisation toolkit, install the BlenderBIM Add-on, and start drafting up an proposal :)

Moult, 2024-03-19:
There will be some integration required with IfcOpenShell as IFC geometry will likely be used as inputs for voxec.

Bundling may be as simple as including the executable cross-platform with the add-on and executed with subprocess, or as complex as building Python bindings to voxec and replacing the DSL.

aothms, 2024-03-19:
With the latest release there is also a python binding: https://github.com/opensourceBIM/voxelization_toolkit/releases/tag/v0.3.2 We currently only have Linux builds. If you're on Windows try WSL.

This will make integration a bit more straightforward.

```python
f = ifcopenshell.open(ifn)
s = ifcopenshell.geom.settings(
    USE_WORLD_COORDS=True,
    DISABLE_OPENING_SUBTRACTIONS=False,
    USE_BREP_DATA=True,
    DISABLE_TRIANGULATION=True
)
it = ifcopenshell.geom.iterator(s, f)
if it.initialize():
    while True:
        elem = it.get()
        geom = elem.geometry.brep_data
        vox = voxec.run('voxelize', geom, method='volume')
        vox_bounds = np.array(vox.world_bounds())
        print(vox_bounds)
        print(vox.count())
        vox.boolean_union(...)
        if not it.next():
            break
```

`voxec.run()` accepts the commands here https://github.com/opensourceBIM/voxelization_toolkit?tab=readme-ov-file#available-commands

this `iterator` loop is something you will also see in the blenderbim codebase to get all geometrical elements out of an IFC BIM model.

With the python bindings it's much more natural do have some proper control flow and integration.

There are some example voxelization scripts here https://github.com/opensourceBIM/voxelization_toolkit/blob/master/python/voxec/__init__.py#L227 Note that this is the **old** way of invoking voxec.exe with a textfile to execute. Within the project we can rewrite these `voxec.run()` command in Python and allow for much more interactive feedback to the user.

I know that @Moult also has some sample scripts to estimate formwork (the molds in which concrete is poured at the construction site).

The python bindings also allow for much more natural integration in terms of visualization. You could draw the voxels immediately to the Blender OpenGL context.

Things are still under development and a bit fluid, but there is really an interesting foundation to build upon to create a new type of analysis interface in CAD applications for the built environment that you can prototype.

ChinmayaBiswal7, 2026-03-10:
Hi @Moult , thanks for the message!

I’d be happy to join a meeting to discuss the project direction with you and @aothms . I’ll join the OSArch Matrix chat so we can coordinate a suitable time there.

At the moment, I’m at university, so I may not be able to join a voice call immediately, but I’m available via text and can schedule a meeting time later today.

Looking forward to the discussion.