opencax/GSoC #91: Geometry Verification and Validation GUI in Qt (AI Project)

brlcad, 2024-02-21:
## Outline

Help develop a new GUI application that checks geometry for common issues and/or helps fix them. 

## Details

A new GUI is in prototype development (built on Arbalest) that checks geometry files for common verification and validation (V&V) issues such as topology errors, solidity errors, and more.  It's very much an experimental work in progress and we'd like your help to make it complete.  The overarching goal of this effort is to extend our prototype in a significant way, either improving usability, checking for more issues, improving the Qt GUI infrastructure integration, integrating workflow(s) for review and repair, or leveraging AI to identify and/or fix issues.

## Expected Outcome

You will propose a complete project description that identifies the specific objectives you'll aim to achieve.  It's expected that you'll leverage the previous work (talk with us to get access to those materials).  The proposal should identify 3-10 primary objectives that are researched and specific, starting with our previous effort.

We essentially want a tool that "compiles" geometry reporting warnings and errors for issues encountered, akin to compiling source code in Visual Studio or Eclipse.  There are questions of application architecture to resolve (e.g., whether to extend 'arbalest', integrate 'qged', integrate 'gist', etc).  We want the tool to be graphical and interactive.  We want it to have the ability to generate reports for auditing.  Some of those capabilities exist in isolation, but none exist as a tool tailor-made for 3D geometry V&V.

## Future Possibilities 

This is a long term priority project with future possibilities in:
- GUI infrastructure
- AI integration
- geometry healing and repair workflows
- geometry auditing
- geometry standards development

### Skills
Qt, C/C++

### Difficulty 
Easy or Medium depending on the objectives

### Size 
long (350h) preferred, but medium (175h) also possible

### Additional Information

* Potential mentor(s): [Sean Morrison](mailto:devs@brlcad.org)
* Organization website: https://brlcad.org
* Communication channels: https://brlcad.zulipchat.com

brlcad, 2024-03-27:
Additional reference material for this project idea is available at [https://brlcad.org/design/v&v/Geometry_V&V.tar.gz](https://brlcad.org/design/v&v/Geometry_V&V.tar.gz)

VW100, 2025-03-19:
Hello,
I am a second-year student proficient in C++. I went through the github page and I am interested in contributing to this project under GSOC 2025.
Can you please guide me and provide more details regarding this project?

Regards,
Vaibhav Wadhwani

drossberg, 2025-03-20:
Hi @VW100, you should have a look at [Geometry_V&V.tar.gz](https://brlcad.org/design/v&v/Geometry_V&V.tar.gz) and join our Zulip chat.

Nanashi-0099, 2025-03-24:
Hello,
I am Hello,
I am a second-year IT student currently learning Data Structures and Algorithms. I am interested in contributing to this project under GSoC 2025.
Could you please provide more details about the project and guide me on how to get started with contributions? I would love to understand the key areas where I can contribute and any initial tasks or issues I should focus on.
Regards,
Danish Kazi

drossberg, 2025-03-27:
Hi @Nanashi-0099, I see that you already joined our Zulip chat. This is the right place to ask questions. And I suppose that you had already a look at [Geometry_V&V.tar.gz](https://brlcad.org/design/v&v/Geometry_V&V.tar.gz) too.