opencax/GSoC #93: Turn BlenderBIM into a client for remote BIM-collaboration on existing OpenCDE-API-server with a graph backend

marwiss, 2024-04-01:
## Outline

The aim is to turn BlenderBIM into a client for remote BIM-collaboration and remote BIM-model-sharing through a Common Data Environment (CDE), using the already developed OpenCDE API server and the OpenCDE API specifications provided by buildingSMART: BCF API and Documents API.

OpenCDE API:s are open standards so the project enable usage of BlenderBIM as a client on all other BIM-servers that implements the OpenCDE API:s.

## Details

An OpenCDE API server that implements all buildingSMART OpenCDE API:s (BCF API, Documents API and Foundation API) has been developed in python and the FastAPI framework. Solibri Office was used as a client for testing the server during its development.

The code of the OpenCDE server is loceted in the IfcOpenShell repository: https://github.com/IfcOpenShell/IfcOpenShell/tree/v0.7.0/src/opencdeserver

The OpenCDE API:s is a set of open API-specifications provided by buildingSMART. https://github.com/buildingSMART/OpenCDE-API

- BIM Collaboration Format API (BCF API) is used for collaboration on shared BIM models through a remote server. BCF API has the same purpose as BCF XML (which is a file format) but the difference is that a server is used for communication instead of files. https://github.com/buildingSMART/BCF-API
- Documents API is used for establishing a server that acts as a Common Data Environment (CDE acc. ISO 19650) sued for sharing models, documents et.c. https://github.com/buildingSMART/documents-API
- Foundation API is used for authentication et.c. and must be implemented by any OpenCDE server. https://github.com/buildingSMART/foundation-API

To summarize: The model (the IFC data) will normally be shared to the server using Documents API, and downloaded form the server using the same API. BCF API can be used for remote collaboration on the models located on the server et.c.

The purpose of the open API specification is to enable independent development of clients and servers that can communication with eachother. The server has already been developed and shared as open source. At the moment there is no open source client with a graphical user interface for the OpenCDE API:s. A python library for BCF API communication is available https://pypi.org/project/bcf-client/

The aim of this project is to turn BlenderBIM into a client (that already have BIM capabilities) that can communicate remotely with (and make use of) the existing open source OpenCDE-server on: https://github.com/IfcOpenShell/IfcOpenShell/tree/v0.7.0/src/opencdeserver

## Expected Outcome

- BlenderBIM is a BCF API client - able to communicate remotely with the already developed OpenCDE-server
- BlenderBIM is a Documents API client - able to communicate remotely with the already developed OpenCDE-server
- User interface in BlenderBIM for remote BIM collaboration using BCF API
- User interface in BlenderBIM for remote model download and sharing using Documents API
- Simplify the process of turning your computer into an OpenCDE-server and inviting colleages to collaborate on your BIM model
- Simplify the process of deploying the OpenCDE-server as a BIM-server to the cloud for remote collaboration with BlenderBIM as a client
- Implement the routes/endpoints of the OpenCDE API specifications that Solibri Office does not support. I.e. implement the full specification using open source server (OpenCDE-server) and open source client (BlenderBIM).

## Future Possibilities 

- Round-tripping of IFC data between IFC STEP (or python or C++ objects in IfcOpenShell) and the OpenCDE server graph DB. More info on storing IFC data as label property graph (LPG) here: https://www.sciencedirect.com/science/article/pii/S0926580523000389
- Potential synergies with the other project "Web-based UI integration with Blender" https://github.com/opencax/GSoC/issues/87 because the Web-based UI could be hosted by the same OpenCDE-server as in this project.
- Visualization of IFC data as graph in that Web-based UI.

## Project Properties

### Skills

- Python, including how to setup a minimal basic local server using FastAPI. https://fastapi.tiangolo.com/
- Blender Python API to develop a BCF API user interface in BlenderBIM.
- Cypher query language and any graph DB that implements Cypher (such as Neo4j or MemGraph). https://neo4j.com/docs/cypher-manual/current/introduction/
- API development.

### Difficulty 

Hard

### Size 

Long (350h)

### Additional Information

* Potential mentor(s): [Martin Wiss]
* Organization website: https://blenderbim.org/ http://ifcopenshell.org/
* Communication channels: [OSArch](https://osarch.org/chat)

Kenyg369, 2024-04-03:
@marwiss 
I am interested!! , 
but the GSoC 2024 application submission is just due after you posted this. 😥

marwiss, 2024-04-03:
I am sorry @Kenyg369 my mistake.

marwiss, 2024-04-11:
@ayushgupta9906 thanks for your interest! Nice that you are enthusiastic. I think the deadline for GSoC is over, but this is open source, so I guess voluntary contributors are welcome at anytime. Maybe it would be a good idea to create a team for this project, because it is quite complex and requires many skills. You can contact @Moult for more information.

YatinSharma37, 2025-01-25:
Title: Enhancing BlenderBIM with OpenCDE Client Capabilities for Remote BIM Collaboration

Personal Information:

Name: Yatin
Institution: Chitkara University, India
Degree Program: Bachelor of Engineering in Computer Science and Engineering (4th Semester)
Interests and Experience:
Strong interest in BIM and neural network projects.
Practical experience with collaborative projects, including working with design companies.
Full marks in Computer Graphics course.
Proficient in Python, FastAPI, and API development.
Synopsis:
This project aims to transform BlenderBIM into a fully functional OpenCDE client. The goal is to enable remote BIM collaboration and model sharing using open APIs developed by buildingSMART. BlenderBIM will leverage the OpenCDE API specifications (BCF API, Documents API, and Foundation API) to facilitate seamless communication with the existing OpenCDE server. The project will focus on developing user interfaces in BlenderBIM, enhancing remote collaboration features, and streamlining server setup for both local and cloud environments.

Benefits to the Community:
This project aligns with the OSArch initiative by promoting open standards and enhancing collaboration in the Architecture, Engineering, and Construction (AEC) industry. By integrating OpenCDE API capabilities, BlenderBIM will become a versatile tool for professionals, bridging the gap between standalone and remote BIM workflows. It will also foster interoperability between independent clients and servers following the buildingSMART specifications.

Deliverables:

Foundation API Integration:

Enable BlenderBIM to connect to OpenCDE servers for authentication and user management.
Add functionality for user registration, invitation, and deletion.
BCF API Client Development:

Implement remote collaboration features for BIM models hosted on OpenCDE servers.
Create a user interface for managing collaboration tasks.
Documents API Client Development:

Facilitate model sharing and downloading between BlenderBIM and the OpenCDE server.
Develop an intuitive user interface for file management.
Server Setup Simplification:

Simplify the process of turning a local machine into an OpenCDE server.
Provide options for deploying the server on cloud platforms.
Optional Deliverables:

Extend server capabilities to support additional OpenCDE API endpoints.
Explore integration with web-based UIs for enhanced visualization and management.
Technical Skills Required:

Python programming (FastAPI for server development).
Blender Python API for UI and plugin development.
Cypher query language for graph database interactions.
API development and integration.
Challenges and Mitigation:

Challenge: Understanding and implementing the OpenCDE API specifications.
Mitigation: Thoroughly study the provided documentation and existing server implementation.
Challenge: Developing intuitive user interfaces in BlenderBIM.
Mitigation: Refer to Blender's UI development guides and collaborate with mentors.
Challenge: Ensuring seamless interoperability between client and server.
Mitigation: Regularly test with the existing OpenCDE server and other API clients like Solibri Office.
Timeline:

Week 1–2: Understand OpenCDE APIs and existing server code. Set up the development environment.
Week 3–6: Develop Foundation API integration and user management features in BlenderBIM.
Week 7–10: Implement BCF API client functionality and collaboration tools.
Week 11–14: Integrate Documents API for model sharing and downloading.
Week 15–16: Simplify server setup and explore cloud deployment options.
Final Weeks: Test, document, and refine all features.
Future Possibilities:

Expand BlenderBIM to support IFC-server capabilities, including IFC data round-tripping with graph databases.
Collaborate with web-based UI integration projects to offer advanced visualization tools.
Extend OpenCDE server capabilities to support additional buildingSMART specifications.
Communication:
I am enthusiastic about collaborating with mentors and the OSArch community through the provided communication channels. Regular updates and discussions will ensure alignment with project goals.

References:

[IfcOpenShell OpenCDE Server Code](https://github.com/IfcOpenShell/IfcOpenShell/tree/v0.7.0/src/opencdeserver)
[buildingSMART OpenCDE API Specifications](https://github.com/buildingSMART/OpenCDE-API)
[FastAPI Framework](https://fastapi.tiangolo.com/)
[BlenderBIM Documentation](https://blenderbim.org/docs/)